gpstar81/GPStar-proton-pack #241: Attenuator doesnt comunicate with pack

ecto1a2003, 2024-02-05:
hi

 i've wired an esp32 into the pack as shown in the documentation. the web-ui shows up but all data is dashes. theres also no blue blinking light. any thoughts?

ecto1a2003, 2024-02-05:
oh, im using the fw that came with the kit currently. i don't have the wand done yet. ive also tried 2 different esp32

gpstar81, 2024-02-05:
If you got your kit after Dec 7, then it should have the firmware that is on the gpstar link above, In that case flash the attenuator firmware from that link. The one in the Preferences branch has more features that are still being worked on, and requires flashing the pack and wand with updated firmware from it as well.

ecto1a2003, 2024-02-05:
got my kit in january so id hope its up to date. ill flash the atenuator again

ecto1a2003, 2024-02-05:
well guess that did it. not sure how i got the wrong fw on the unit. thanks!

closed 2024-02-05 by ecto1a2003