gpstar81/GPStar-proton-pack #266: Sound distortions

Toy203, 2024-03-21:
I got some weird distorsions on 1984/89 sounds bite for boot and shutdown. I've turned down my ampli for the second test and removed one of the two speaker since i dented it so i though that might be the cause, but still i can hear the distortions.

nomakewan, 2024-03-21:
So, a few questions if you don't mind.

1. Do you have a ground loop isolator between the WAV Trigger output and the speaker amplifier? Not having one can cause distortion to appear in the output audio.
2. The video link to "I can hear a hiss when music track starts playing" leads to a "This video isn't available anymore" page.

A lack of ground loop isolator can be the cause for the distortion effects seen in the first, second, and fifth videos you linked. The distortion in the third video you linked is actually in the audio file itself, which you can hear here: [Shutdown Sound](https://github.com/gpstar81/GPStar-proton-pack/raw/main/sounds/002_shutdown.wav)

Fixing that distortion should be easy but I'll have to ask the other devs if it was in the sound file intentionally since it's been like that for well over a year.

Toy203, 2024-03-21:
Sorry i forgot to mention, yes this is with ground loop isolator installed. I'm testing the GPstar audio board atm so i figured the default amplification might be a bit high.

Here's the hiss video link again
https://youtube.com/shorts/d2VHScUqNOk

Thanks for letting me know that weird wishy washy sound it's in the file though, i never noticed. So weird!

The fifth video was just to remark it feels like the pack and the wand aren't perfectly synched up with audio in my case. Maybe it's just me though. I'm becoming Lord Nitpick.

nomakewan, 2024-03-21:
Ah, I wasn't aware you were testing the GPStar board! In that case this could indeed be related to that code branch, my apologies. Thank you very much for being a beta tester! And yes, that video link now works.

Hey, being Lord Nitpick is awesome, because it makes the kit that much better for everyone when these sorts of things get found and fixed.

gpstar81, 2024-03-21:
Thanks for the wand sound timing video. That confirms my suspicions on a few things where I need to fix up a bit more which may solve a few other small timing issues that we discussed earlier.

Could you also point out what the volume levels are on both the pack crank knob and the amp, to give me a idea on a few other things.

Thanks.

Toy203, 2024-03-22:
> Could you also point out what the volume levels are on both the pack crank knob and the amp, to give me a idea on a few other things.
Not sure i get this. I keep my amp knob on half way through. Pack don't have a fixed setting i tried it at max and lowered it to make some videos.

nomakewan, 2024-03-23:
Just a quick mention that the latest updates to the Audio WIP branch includes a new shutdown sound to remove the distortion heard there. Thank you for your report!

closed 2024-04-11 by Toy203