gpstar81/GPStar-proton-pack #276: Overheat problems

Toy203, 2024-04-11:
Just updated to the latest revision and cleaned EEPROM. Overheat function is enabled in the eeprom. 

If i fire the wand on power level one (with all levers up), overheat sequence is triggered immediately . The wand shoots for about 1 second then the pack enter overheat: alarm sounds trigger for way too short time (it only does a single beep, i believe is "034_pack_beeping.wav" or "060_beep_8.wav"), lights slow down, strobo in the n filter starts and smoke hiss is heard.

I cannot seem to be able to trigger overheat while firing with the other power levels. I used a c clamp on the intensify just to be sure, overheat doesn't seems to trigger on continuos firing. 

If i rapidly press intensify thought, it works. On level 5 if i press it 6 times overheat is triggered, but it's a weird quick overheat without the ramp down etc. Just an alarm sounds is played for a couple of seconds (it sounds like "067_beeps_low.wav") and no smoke hiss is heard or lights in the n filter.

gpstar81, 2024-04-11:
I had same issue. A power reset was required after wiping the config-EEPROM then re-configuring it again if required.

Toy203, 2024-04-11:
Power cycle seems to have fixed most of the issues. 

I just noticed the bargraph animation defaults to the regular animation before shutting down due to overheat, which is a bit clunky coming from the super hero firing animation. Lord Nitpick is back in town.

https://github.com/gpstar81/GPStar-proton-pack/assets/153134073/513cdf2e-adbd-4939-aed0-b88ec9f0fe1f

nomakewan, 2024-04-11:
> Power cycle seems to have fixed most of the issues.
> 
> I just noticed the bargraph animation defaults to the regular animation before shutting down due to overheat, which is a bit clunky coming from the super hero firing animation. Lord Nitpick is back in town.
> 20240411_120253.mp4
> 
> Also, i still can trigger that weird alarm status if i rapidly press Intensify while firing. I can find a certain over-sensibility on the intensify button too, firing state switch off easily if i wiggle the thumb a bit while pressing it. Might be my mighty soldering but didn't have this feeling before.

The bargraph animation is actually from the video game for overheat, and has been this way since the original firmware release that implemented overheating. As such it's actually intended behavior as of now. Do you have a suggestion for a change to the overheat animation in Super Hero mode?

The alarm state is intentional and was implemented a while back to prevent people from overloading the serial bus by making rapid firing changes. This is the "anti button mash" system.

As for the oversensitive Intensify button, that is very odd indeed; if anything the button should be slightly _less_ sensitive now that we've switched button libraries on the wand. I also cannot replicate that behavior on my bench setup that's using stock Hasbro switches (holding Intensify then wiggling my thumb on the button while doing so).

nomakewan, 2024-04-11:
Thinking about it, I'm going to just implement an EEPROM_VERSION into the EEPROM system so that it can handle that reset correctly when we make changes to the EEPROM objects. Yes, it means we'll have to remember to update that version when making changes to the EEPROM object(s) but a little bit of responsibility on the back-end will alleviate a lot of potential user headache.

Toy203, 2024-04-13:
Will update as soon as I can. In the meantime, on the new quick vent feature I got no smoke action. Smoke works well on regular overheat sequence, but the quick vent doesn't trigger my Wasteland smoke machine. Got light and (I think) fan action but no smoke.

nomakewan, 2024-04-13:
As soon as I'm done fixing a few other things I noticed while working on things yesterday I'll take another look at Quick Vent. Most of the smoke-related code was copied straight out of the overheat parts, but there's of course always the chance that I placed something on the wrong side of a curly brace. Thank you for the report!

Toy203, 2024-04-16:
Is the new quick vent still tied to the Overheat timing as it was before? I ever only used it at lvl 5 and with increased timing (think i used 12 secs or so) for the  smoke to be really noticeable on my setup, and i liked the effect. Or will the new quick vent be adjustable like Overheat?

nomakewan, 2024-04-16:
> Is the new quick vent still tied to the Overheat timing as it was before? I ever only used it at lvl 5 and with increased timing (think i used 12 secs or so) for the smoke to be really noticeable on my setup, and i liked the effect. Or will the new quick vent be adjustable like Overheat?

The new quick vent is tied to the overheat timing, yes. In the video game, a full overheat lasted 2 seconds, while the quick vent lasted 1 second. As such, I have coded Quick Vent to last 1/2 the duration of a full overheat or 2 seconds, whichever is longer.

Toy203, 2024-04-17:
Still got  no smoke on quick vent at all. I can see my Booster tube led lights up (it's directly connected to the Gpstar mobo) but the rest of the Wasteland kit doesn't trigger.

Toy203, 2024-04-17:
Will check asap. I can definitely hear the smoke pump go on regular Overheat (18 seconds)

nomakewan, 2024-04-17:
Okay, so despite calling the function that triggers smoke, smoke is not being triggered even when the timer for smoke is not in use (overheat sync to fan disables the timer that would otherwise have been dedicated for smoke and simply tells it to run as long as quick vent lasts).

I'll take a look at the code again to see if I can figure out why. Maybe there's a check in the smoke function itself I missed.

nomakewan, 2024-04-17:
New update pushed, give that a shot!

Toy203, 2024-04-17:
Got smoke! But no fan action.

nomakewan, 2024-04-17:
D'oh...same problem in the fan function. Gosh darn it, silly checks. One moment.

Toy203, 2024-04-17:
BTW I saw no eeprom erasing with the latest update. Maybe I got it wrong, do I still need to power cycle it?

nomakewan, 2024-04-17:
If you already tried the firmware that was in the pull request before it got merged, then that's normal. The EEPROM will only reset if you flash a firmware that changes how the EEPROM works (for example, we add new features to the menus). Since the last several updates didn't touch the EEPROM, it won't clear it either.

Toy203, 2024-04-17:
Updated, got no movement from the booster tube fan i've connected to the gpstar board. Wasteland kit nfilter fan and led all works perfectly. Tried both with fan synch enabled and disabled.

nomakewan, 2024-04-17:
That's because I did not enable the booster tube for quick vent, intentionally, for quick vent.

Toy203, 2024-04-17:
I'd personally prefer all the bell and whistle to go for sure, i used the Overheat wand function a lot while roaming on the floor and the booster tube tend to trap the smoke inside due to gravity. I'll leave it to the experts though! =)

nomakewan, 2024-04-17:
Update pushed, booster tube is go.

Toy203, 2024-04-17:
Smoke galore it is!

Toy203, 2024-04-17:
Nope still no fan movement in booster tube

nomakewan, 2024-04-17:
AHHHH COPY AND PASTE

remember kids: do not code before you have finished your morning caffeine

Toy203, 2024-04-17:
Smoke galore it is!