gpstar81/GPStar-proton-pack #277: Random Led flashing on lid connect

Toy203, 2024-04-11:
As per title, i saw some random green flashing when connecting the cyclotron lid to the body, see video. I'm running latest firmware available and Frutto 5 leds pucks. Also does it randomly with the pack on.
https://youtube.com/shorts/FBLkAETlCY4?feature=share

nomakewan, 2024-04-11:
I think I know the problem. Previously as long as the pack was off it was firing off a reset command every single tick of the microcontroller which was masking a lot of sins. Now that that bug has been fixed, all the bugs that were hidden by it are coming out.

I'll take a peek at the code that handles the cyclotron lid detection as soon as I've had breakfast. Thank you for the report!

nomakewan, 2024-04-11:
@Toy203 Please check PR #278 and see if it resolves this issue. Note that both the wand and pack will need to be flashed, and that the first time you flash it will reset all EEPROM settings to default (by design, so that you do not have to clear settings and power cycle the system manually after a firmware update).

Toy203, 2024-04-16:
Sorry, i needed the pack functional for promotion so far. Will update as soon as possible!

nomakewan, 2024-04-16:
Well with any luck the new PR will be even more stable and functional than previous. Still, completely understandable not loading experimental code right before an event. Thank you again!

Toy203, 2024-04-16:
You guys are going at a mind blowing pace!

closed 2024-04-16 by Toy203