gpstar81/GPStar-proton-pack #279: Audio weirdness

Toy203, 2024-04-17:
Just updated the latest version. I got this weird status when i control SFX volume from the wand menu during operations. Once it maxes out it bring the wand loop wooshing very loud on top of the pack idle. This feels weird as the sound it's not usually so prominent in the mix. See video attached:
https://youtu.be/_uZB8sRb00A

I can hear the new Afterlife shutdown (348_pack_shutdown_afterlife_alt.wav) which is awesome and bass-y (althought the tail of the last 2-3 seconds with the electric sounds feels very long and adds to an already long shutdown sequence), just not sure how the pack decides what version to play on shutdown.

gpstar81, 2024-04-17:
Do you have the extra wand sounds enabled which makes the pack play the wand idle sounds?

For the new shutdown, it is the same length as before, just 2 new sound effects added depending on how you start or shutdown. When using the ion arm switch, there is a extra sound coming from the pack of flipping a large switch, the same when shutting down via the ion arm. When using the wand to start or shut down, the switch sound is not played.

nomakewan, 2024-04-17:
Extra Wand Sounds is enabled by default, so when the EEPROM reset it would have been enabled.

Toy203, 2024-04-17:
> Do you have the extra wand sounds enabled which makes the pack play the wand idle sounds?
Yes indeed it's enabled.

Omg i get it now the new sound idea is just the cherry on the nuclear cake guys.

nomakewan, 2024-04-17:
I agree with shortening the shutdown sound, but also keep in mind that you'd want to likewise shorten the shutdown lighting effects. I think right now the audio is synchronized to the ramp down on the power cell and cyclotron LEDs. So shortening the sound is more than just...well, shortening the sound.

Toy203, 2024-04-17:
Yes I meant shortening the whole thing just a tiny bit would do.

closed 2024-05-01 by Toy203