gpstar81/GPStar-proton-pack #280: Barrel LED Stream Effect not ending properly

nomakewan, 2024-04-19:
This bug has actually been in the code forever, but it became more obvious now that we have a high-resolution barrel LED solution. Visually, this bug causes the 'spark' effect of a stream (or the entire effect in the case of Meson mode) to get 'stuck' in the barrel the moment the user takes their thumb off of the Intensify switch, rather than the animation completing up to the end of the barrel naturally.

This bug is caused by the stream effect being controlled by `modeFiring()` being called every tick while firing, which then calls `fireStreamEffect()` that operates on a loop of the `ms_firing_stream_effects` timer. Because `modeFiring()` doesn't get called unless firing, this causes any stream effect that was in progress to "stick" in whatever LED position it was last in.

The reason this was so hard to notice before is because `fireStreamEnd()`, controlled by the `ms_firing_lights_end` timer which is started the moment firing ends, "runs up the barrel" from base to tip setting each individual LED along the way to off, which of course unsticks the stuck LED once it reaches that point by turning it off.

What should happen is if a stream firing effect is currently visible anywhere in the LED array, then its animation should always be allowed to complete (reach the end of the barrel and then be turned off) no matter what. If no stream effect is present the instant the user releases Intensify or shuts off the wand, nothing happens. In order to make sure the "end stream" effect does not outpace/underpace this effect, `ms_firing_lights_end` should always be matched to the speed of `ms_firing_stream_effects` or be slightly slower (so that the stream effect animation always completes before `fireStreamEnd()` sets the LEDs to off).

I did try my hand at fixing it myself, but my experiments so far haven't gone as planned. I don't quite have my head wrapped around how the barrel LED animation system works, so I would appreciate assistance in resolving this issue. Thank you!

gpstar81, 2024-04-19:
I know about it. I still have more animation work to do and will get around to it.

gpstar81, 2024-04-19:
Thinking about this more, the entire system will need to be re-done with a tracking system that tracks each fire event and led status for that event to let the animation play out, versus the current method of wiping out each led when firing ends. It will take a bit of work to do. Perhaps best to do it not for this next release (v5) but for the release afterwards.

nomakewan, 2024-04-19:
Agreed; when I was looking at the code it seems the current version tracks the currently-edited LED with a single variable, which of course makes it difficult if not impossible to have the animations end the "right" way.

I have an idea for a stopgap measure but I'll need to do some testing. If I can get it to work and it isn't terribly out of sync with the firing tail sound effects I'll see about having that for v5. Worst case fully agreed, animation and menu refactors should come after v5 is released.

nomakewan, 2024-04-27:
Fixed in commit [5e2c1f3](https://github.com/gpstar81/GPStar-proton-pack/commit/5e2c1f3b11e2d8071a8e5e7efdec3366d4df4542).

closed 2024-04-27 by nomakewan