gpstar81/GPStar-proton-pack #282: Random issues and comments

Toy203, 2024-04-22:
Just flashed latest software:

In Eeprom menu, i got few messages playing on the wand speaker but not on the pack, specifically:

36 led cyclo sound bite
Wand beeping disable

Also it feels like in both AF and FE this latest setting is not really read correctly, it is still beeping in loop dunno if it was intended. I got wand sound routed to the pack.

The Eeprom still reports the option to light up 1 ore 3 leds for 84/89 mode, but we'll have 5 eventually isn't it the case?

gpstar81, 2024-04-22:
The 84/89 1 or 3 LED thing was a old feature request from somebody a year ago as some people use diffusers and want the extra light. Even with the 5 led and 9 led options, it will stay for now at 1 or 3 depending on the setting.

Wand sounds routed to the pack is the new default setting now, and needs to be disabled if you do not want that in the EEPROM.

I will check the 36 led and wand beeping disable sound bytes to make sure things are ok. Both SD cards were updated?

Toy203, 2024-04-22:
Yes both SD with the same sound files.

Was the AF shutdown audio changed as well? Now it doent's sounds like a dying engine anymore and i feel something is missing lol.

gpstar81, 2024-04-22:
As far as I know, there have been no changes to the shutdown sound since over a week ago.

gpstar81, 2024-04-22:
Correction, it looks like the sound was modified and merged in a few hours ago.

Toy203, 2024-04-22:
My pack isn't currently playing 348_pack_shutdown_afterlife_alt or 036_pack_shutdown_afterlife.wav for sure, just the wand shutoff 017_wand_shutdown.wav .
So weird

nomakewan, 2024-04-22:
> My pack isn't currently playing 348_pack_shutdown_afterlife_alt or 036_pack_shutdown_afterlife.wav for sure, just the wand shutoff 017_wand_shutdown.wav . So weird

The check here is insanely simple. If SYSTEM_YEAR == SYSTEM_AFTERLIFE or SYSTEM_FROZEN_EMPIRE, then when packShutdown() is called, it will be commanded to play S_PACK_SHUTDOWN_AFTERLIFE_ALT.

If SYSTEM_YEAR is SYSTEM_1984 or SYSTEM_1989, then instead it will play S_SHUTDOWN and either S_PACK_SHUTDOWN or S_GB2_PACK_OFF depending on whether it's 1984 or 1989.

If the pack is only playing 017_wand_shutdown and not any of the above, it sounds like packShutdown() isn't being called.

Toy203, 2024-04-22:
SD card was the problem, i've recopied the file once more and now i got sound correctly.

Toy203, 2024-04-22:
017_wand_shutdown

closed 2024-04-22 by Toy203