gpstar81/GPStar-proton-pack #323: Wand vibration settings not kept after power cycle

Toy203, 2024-05-31:
The wand vibration settings is kept off regardless of the eeprom settings when the pack battery is turned off and on. I'm on 5.1.2 and can reproduce the issue. Need to get into the eeprom settings again to activate vibration, regardless of the hard switch position.

nomakewan, 2024-06-01:
Note this is now the second report of the same exact issue, as we had a user on the FB group report the same.

Is this only on 5.2.1 or does it happen on older firmwares as well?

Toy203, 2024-06-01:
Unfortunately I couldn't test on previous firmware so far.

nomakewan, 2024-06-01:
Okay. I will check over the code again to see if there is something obvious. I don't recall us touching any of the vibration-related functions in 5.2.1 but always worth another look.

I will also ask the other user if they are using an Attenuator, as I know you are. If you don't mind, could you test removing your Attenuator cable, then setting vibration options, then power cycling without the Attenuator connected to see if the same bug still occurs?

Toy203, 2024-06-01:
I don't have an attenuator installed, only the ESP inside the pack actually.

nomakewan, 2024-06-01:
My mistake, though they operate identically I understand it would be much more difficult for you to test removing it since you would have to reopen the pack. I will stick to just reading the code to try to locate the issue, then. Thank you.

nomakewan, 2024-06-01:
I have located some potential issues with how vibration is handled, though those issues did not start in 5.2.1. Perhaps they just weren't noticed until now.

Will include these fixes in 5.2.2.

Toy203, 2024-06-02:
Fixed with 5.2.2

closed 2024-06-02 by Toy203