gpstar81/GPStar-proton-pack #326: Quick vent pump stops

Toy203, 2024-06-08:
I've set a longer overheat duration for level 5, 30 secs, since I plan to use quick vent  when I roam around. . The quick vent feature correctly turns on the leds for the half time duration as expected when engaged, but the pump stops after 4-5 seconds.

nomakewan, 2024-06-10:
The pump stopping after 4 seconds is intentional. The bug is actually that your quick vent does a duration longer than 4 seconds.

I see where the bug is (it was just a reversed inline conditional while setting the timer) and this will be corrected for 5.3.0.

Toy203, 2024-06-10:
So i'll still be able to set up a longer overheat sequence to get half of that duration quick vent feature, correct?

nomakewan, 2024-06-10:
That was not the intention of the video game quick vent feature. It was just supposed to be a rapid vent like in the game, and so the original intent was to have it top out at 4 seconds.

For people who just wanted to show off smoke, that's why the "long press" quick vent was implemented that forces the full overheat sequence.

Toy203, 2024-06-10:
Ah i see, this is a bit unfortunate. I really preferred to be able to trigger just the smoke effects with custom duration while roaming around during events, without triggering all the overheat alarms etc.

nomakewan, 2024-06-10:
Well, it can always be changed. If someone's requesting it, I have no reason to reject that change request. It's a bug either way, just depends on which side of the calculation you want to call wrong.

I'll change it so it's duration / 2 on everything for you.

closed 2024-06-10 by Toy203