gpstar81/GPStar-proton-pack #333: Cyclotron lid slime mode not working

Rizzbuster, 2024-07-08:
Cycling lid issue. Testes all the year modes and slime mode is not working. I did get continuing fire at one point but can't pin point how it happened yet.

Toy203, 2024-07-08:
Can reproduce, no lights in the circles on Slime mode. I got 9 led pucks mounted.

DustinGrau, 2024-07-08:
From reports, one user is on the stock lights, another on Frutto, so this is not immediately related to RGB vs. standard.

nomakewan, 2024-07-08:
Found the issue. It's so stupid as to be unreal.

nomakewan, 2024-07-08:
So to fix the issues with ramp speeds not being correct if you boot the Proton Pack in Slime and then switch firing modes (or recover from overheat in slime and switch firing modes), I moved the call to the slime lights into the actual cyclotron control loops. The problem is I forgot there was a call that _skips those control loops entirely_ if we're in slime mode.

Fixed now.

nomakewan, 2024-07-08:
> Can reproduce, no lights in the circles on Slime mode. I got 9 led pucks mounted. Also, in Slime mode (AF) my power cell is still running once i turn off the pack from the ion arm or wand.

Please test the [new binary](https://github.com/gpstar81/GPStar-proton-pack/blob/v5.3.1/develop/binaries/pack/ProtonPack.hex) on your pack to see if it resolves all of these issues. If it does, awesome. If not, that lets us know to keep looking into the power cell and vibration stuff.

Toy203, 2024-07-08:
Sorry, no luck. Same behaviour as before both for cyclotron ring and power cell

nomakewan, 2024-07-08:
> Sorry, no luck. Same behaviour as before both for cyclotron ring and power cell. Should i erase eeprom and try?

Erasing EEPROM has not been necessary since 5.0.0. It will have no effect here since the issue is with code, not EEPROM settings.

I will see what else I can do to work this out.

nomakewan, 2024-07-08:
> Sorry, no luck. Same behaviour as before both for cyclotron ring and power cell. Should i erase eeprom and try?

Okay, new binary is pushed (it has the same URL as above). Please test this one.

Toy203, 2024-07-08:
Looks like both problem are fixed with this! Tried also in 84/89, no issues. Cyclotron lights up and no more power cell going after shutdown.

nomakewan, 2024-07-08:
@Rizzbuster See if everything is good on your end as well. If so, we can consider this issue Busted!

closed 2024-07-08 by DustinGrau