gpstar81/GPStar-proton-pack #338: [Feature]: Possessor mode

Toy203, 2024-07-08:
### What would you like to add in terms of software changes?

Adding a pack state that resembles it being possessed by the Possessor ghost: cyclotron all lit red pulsing on and off (no rotation) and red power cell color. Growling instead of regular pack sounds. Could be activated like the spectral mode manually or even as an easter egg, although some folks don't like the lack of control on these features me included. 

### Would this request involve any specific hardware?

It'll work better with rgb power cell leds. 

### Homework Completed

- [X] I did my part!

nomakewan, 2024-07-09:
I actually would find this interesting, though on the wand it's a little bit more tricky:

[image]

In the film, all of the indicator lights on the wand turn red when the Possessor takes over. However, at least for now, the only addressable LEDs on the wand are in the barrel, while all other LEDs in the wand are non-addressable single-color LEDs. So the effect would not be nearly as impactful as it was on screen unless we fundamentally change how the wand hardware works, which considering the PCB is already set in stone, would be a tough ask.

Toy203, 2024-07-09:
I was thinking it with the pack in display, so the wand should be resting. I think just having the leds blink or all turn on static would be a close enough look.

Toy203, 2024-08-07:
As an easter egg would be fun, but it'll be best to have a trigger for it somehow. It could follow the Spectral Modes although i feel we already have a lot of the platter on that and you'd be rotating for a while...

nomakewan, 2024-08-07:
It shouldn't be a firing mode. If it's random that's easy enough to handle. A trigger however would need to be carefully considered since we only have so many controls available, and need to consider how those controls interact in the various potential operating states of the wand.

In the film, the Proton Pack cyclotron at least continued to spin as normal (though it was already red, so hey). Couldn't see the Power Cell but would be totally willing to make it red. Growling sounds makes sense, as does vibration to go along with those sounds. Likewise having it randomly fire makes total sense (similar to how the proposed "demo feature mode" would work).

Still, it's a ways off. Plenty of time to work things out.

DustinGrau, 2024-09-11:
Adding a note that this could be considered with #367 as another animation/effect to add as an option in the system, but would need to coincide with a refactoring of the cyclotron animation code.

The-Y33TER, 2025-01-14:
all my indicators are red :O

DustinGrau, 2025-10-05:
This is quite an older request (sorry about that) but now that the GPStar II boards are in production and that requires the use of the RGB Vent Light kit (which changes both the vent and top LEDs) this feature is looking more likely to be possible as the hardware is now available and required for newer kits.