gpstar81/GPStar-proton-pack #339: Power intensityis backwards [Bug]: <short description>

Cevans9820, 2024-07-08:
### What actions were attempted?

I turned on my pack, ans increase the power level on my wand and when I did the power level is backwards, when I fire on level 5 it fires like it's on level one and vice versa.

### What actually happened?

When I move the wand to level 5 to overheat the wand it fired like it was in level 1 

### What was expected?

I wasn't expecting the power levels to be backwards

### Firmware Version

5.3.1

### Mode in Use

Super Hero / Video Game

### Notable Hardware

N/a

### Homework Completed

- [X] I did my part!

nomakewan, 2024-07-08:
If you are referring to the color scheme of the proton stream, that is actually completely intentional.

Here is the team firing their Proton Packs indoors:
[image]

And here is what it looked like after Egon told everyone, "Full Neutronas":
[image]

Which is why in the changelog for 5.3.0 you will find the line:
`-1989 proton stream sounds and colors updated to match film`

Is this indeed what you are referring to?

Cevans9820, 2024-07-09:
> Ghostbusters.II.1989.2160p.BluRay.x265.10bit.HDR.Tigole.mkv_snapshot_00.34.55.601.jpg
> Ghostbusters.II.1989.2160p.BluRay.x265.10bit.HDR.Tigole.mkv_snapshot_01.25.01.301.jpg

Ahhhhh gotcha

nomakewan, 2024-07-09:
Since that seems like what this issue is referring to, I'll be closing it. Feel free to reopen it if you feel there is another bug relating to wand power level in 1989 year mode.

closed 2024-07-09 by nomakewan