gpstar81/GPStar-proton-pack #340: [Bug]: 1989 Proton Stream Power Level Animations inverted

JMills75, 2024-07-09:
### What actions were attempted?

1. Turn on the Proton Pack
2. Set the Year to 1989
3. Fire Proton Stream in power levels 1 through 5

### What actually happened?

The colors appear inverted based on the power level:
**Example**: Power Level 5 is a very deep red, while Power Level 1 is a very bright orange

### What was expected?

The coloring starts at a deeper red at power level 1 and gets progressively more orange as you increase the power level. 

### Firmware Version

5.3.1

### Mode in Use

Any / Both Modes

### Notable Hardware

49 Barrel LEDs

### Homework Completed

- [X] I did my part!

Toy203, 2024-07-09:
It was intended and updated in 5.3, see here:
https://github.com/gpstar81/GPStar-proton-pack/issues/339

JMills75, 2024-07-09:
Duplicate of issue 339.  Closing issue

JMills75, 2024-07-09:
> It was intended and updated in 5.3, see here: #339

Thanks.   I closed the issue and marked is as a duplicate