gpstar81/GPStar-proton-pack #341: [Feature]: Frozen Empire Lockup

DustinGrau, 2024-07-09:
### What would you like to add in terms of software changes?

Putting this out there for visibility: when in the Frozen Empire theme, button mashing on the wand would instead cause the pack to freeze up like in the fight with Garraka instead of simply going into an alarm mode. We've already got some of the effects in place (statis mode ice crackle) but we need the sound effect of the cyclotron grinding to a halt.

Required: The effect would basically halt the spin of the cyclotron lights and then fade out with the ice-up sound.
Possible: When the pack turns back on it would ideally have some sort of ice-breaking sound before powering up.
Optional: Perhaps even during the time that the pack is "disabled" the Garraka chant may be heard in the background.

### Would this request involve any specific hardware?

None required

### Homework Completed

- [X] I did my part!

DustinGrau, 2024-07-09:
I'll be taking this on for the v6 release, though if anyone sees this and can assist with the sound effects needed I would appreciate it!

Toy203, 2024-07-09:
This sounds incredible cool (pun intended)!

Aces6469, 2024-07-10:
I'm not exactly an audio wizard, but I decided to try my hand at the initial freeze up sound
[freeze up.zip](https://github.com/user-attachments/files/16159894/freeze.up.zip)

DustinGrau, 2024-07-10:
> I'm not exactly an audio wizard, but I decided to try my hand at the initial freeze up sound [freeze up.zip](https://github.com/user-attachments/files/16159894/freeze.up.zip)

That's damn near close enough to get started!

DustinGrau, 2024-08-11:
I've begun the work on this and have added code to differentiate the theme (year) mode and play the expected sounds. However, I need to work on the logic which actually halts the cyclotron and pack for the "freeze up" itself. Otherwise, I've got the timing adjusted and new effects working.

DustinGrau, 2024-08-18:
Work continues. Sounds are good, sequence start/end looks good, just working on some of the lighting changes. I'm trying to get the lights to offer a slow-to-stop type sequence but it won't be 100% on this first pass. My goal is to get _something_ out in the near term and expect that we'll clean it up in V6 when we tackle some of the timer logic. There's some other factors that may not be completely correct, such as powering down the wand mid-freeze-up seems to cause the device to go into the full shutdown sequence rather than just go straight for an immediate shutdown.

DustinGrau, 2024-08-21:
This is Good Enough(tm) for a release where we can get comments and critiques. All the desired functionality is there, it just needs some polish which we can get to over time.

DustinGrau, 2024-08-21:
This will be part of the 5.3.3 release

closed 2024-08-21 by DustinGrau