gpstar81/GPStar-proton-pack #343: Possible bug, or new feature that wasn't listed from what I can tell.

Aces6469, 2024-07-10:
### What do you need assistance with?

The startup speed/animation in general on the cyclotron cake seems to be different in frozen empire mode after I installed the new version. I've attached two videos comparing it and and the previous one. I'm not certain if it's just a new feature I didn't spot in the change log, or if it's a bug. Thanks for the insight! 

(if they're labeled wrong, the one that has a longer animation and isn't on it's stand is the one from 5.3.1)

(5.2.2)
https://github.com/gpstar81/GPStar-proton-pack/assets/175177905/fc921599-0125-4e15-8c5c-ac4b8eea7838

(5.3.1)
https://github.com/gpstar81/GPStar-proton-pack/assets/175177905/e6c3aaaa-dac0-4bc2-aec1-c285b910a44f

### Firmware Version

5.3.1

### Homework Completed

- [X] I did my part!

nomakewan, 2024-07-10:
While there was indeed a change to how some bootup worked on the pack, I don't recall touching the ramp for the inner cyclotron. I did find a bug that slipped through though (or rather, a feature that was intended to be included in 5.3.0 but which was accidentally left out), so while fixing that I will further inspect the inner cyclotron code to see if anything jumps out at me.

Thank you for your report!

Aces6469, 2024-07-10:
Just checked it, happens with all modes. Only in FE's era setting though. I'm gonna double check my bug list with a buddy of mine, and see if I just need to reflash.

nomakewan, 2024-07-10:
Okay, I believe I have found where the code was changed that is causing this difference in inner cyclotron ramp up speed. I have not yet determined the reason, but I would like to test my theory.

Are you active on the Facebook group and would be willing to accept a direct message? If not, do you have another preferred method of contact (e-mail, Discord, etc)?

Toy203, 2024-07-10:
I could test it tomorrow morning (Italy time) if you need.

Aces6469, 2024-07-10:
yeah, TJ on the fb group.

nomakewan, 2024-07-10:
Message sent! You may have to check your message requests since it appears we've never talked directly before on FB.

nomakewan, 2024-07-10:
Okay, was able to confirm the cause and the fix, and it is now implemented for 5.3.2. If you have the "brass pack" configuration, a faster ramp will occur to more closely match the sound effects (and how it looked in 5.2.2).

closed 2024-07-10 by nomakewan