gpstar81/GPStar-proton-pack #344: Frozen Empire Lid Off Sound not persisting through spectral modes.

Aces6469, 2024-07-10:
### What actions were attempted?

Switched on everything as normal, lid off like usual. Started up, the ramping sound was there. Swapped to spectral mode, and the ramping effect you usually hear in FE mode with the lid off was missing. In my case, the sound is swapped for an idle, as I changed the sfx to set the whole FE mode as a Brass Plated pack. I use the spectral modes to get the stream green, and to edit the color of my spark effects around the cake, so it kinda busted me up a bit (Pun intended). Either way, here's some footage to help get my point across. Thanks!

https://github.com/gpstar81/GPStar-proton-pack/assets/175177905/40aeec2f-42ea-4eb7-8b2f-8506209958b1

(when I say ramping effect, I'm referring to 334_pack_frozen_empire_boot_effect)

### What actually happened?

Swapped to spectral modes, and '334_pack_frozen_empire_boot_effect' didn't continue to play like it was supposed to (with the lid off).

### What was expected?

'334_pack_frozen_empire_boot_effect' Should still continue to play while in the spectral modes in frozen empire mode with the lid off.

### Firmware Version

5.3.1

### Mode in Use

Super Hero / Video Game

### Notable Hardware

GPStar Supreme Contents, Recommended SD cards + Speakers, and WAV Triggers.

### Homework Completed

- [X] I did my part!

nomakewan, 2024-07-10:
Since this was done intentionally it is not a bug. By design, the special FE sounds only play in Proton Mode.

However, I can consider allowing them for Spectral Custom in the next release.

Aces6469, 2024-07-10:
It wouldn't go unappreciated, using the CTS setup just doesn't do it for me lol. Thanks for the help nomake :)

nomakewan, 2024-07-10:
This is now implemented for 5.3.2.

closed 2024-07-10 by nomakewan