gpstar81/GPStar-proton-pack #345: Frutto LED Panel Colors

Aces6469, 2024-07-13:
### What do you need assistance with?

Apologies if this has been asked and answered before, but does the new inner cyclotron panel lights have a setting where it just sticks to the OG coloring? I'm thinking about yoinking it to reduce the wire density in that area of my pack (mostly because I wanna do the FE D-SUB connection on the clamp).

### Firmware Version

5.3.1 (test version from Nomake with the FE cyclotron fix)

### Homework Completed

- [X] I did my part!

nomakewan, 2024-07-13:
Define "OG coloring," if you could.

Aces6469, 2024-07-13:
should have been more specific in hindsight lol, my b. The Screen accurate Cliplite coloring and orientation. Basically the color scheme the base inner cyclotron kit has with the separate LEDs. The color change upon mode swap is cool and all, but I'm a sucker for the classic look.
[image]

nomakewan, 2024-07-13:
It does not currently have such a setting, but there's no reason why it can't be introduced in v6.

DustinGrau, 2024-08-10:
If I were to read this another way, the feature being requested is to offer a menu/preference option to simply use a plain white for the "on" state for any panel lights, so that the tinted cliplites show their original color. Basically, just ignore any color information for the new RGB LEDs?

nomakewan, 2024-08-10:
That would not be a good idea; unlike incandescent bulbs, LEDs have sharp cutoffs for the spectra they emit. So for proper color representation behind a colored lens, you have to match the color of the lens as closely as you can.

DustinGrau, 2024-08-10:
Got it, so the option will force the LEDs to just use a preset color for the respective position, and won't shift based on the stream selected?

nomakewan, 2024-08-10:
Exactly. For folks who have the prop-accurate lenses installed, the Frutto panel will simply stay locked to the colors of those lenses no matter how you change firing mode or overheat or anything else.

DustinGrau, 2024-08-10:
So for consistency, will we just offer a "VG" toggle for this component like the cyclotron lid and powercell? If the toggle is enabled then the colors shift, otherwise they remain "stock"? I think the current wand menu option cycles through the devices but we may need to redesign that which would be a v6 thing for sure. If I could make an Attenuator preference that could be done pretty quickly.

nomakewan, 2024-08-10:
For v6 it would be an enum probably, with the options for the Frutto panel being Disabled/Single Color/Full Color. Because we don't need a bool to tell if it's installed or not if we're already defining how it should be used.

DustinGrau, 2024-08-10:
Yeah I like the ENUM idea. Maybe "static color" vs "dynamic color" is more apt given the usage.

DustinGrau, 2024-08-10:
I'm starting on refactoring the bool flag to be an enum, just to get us in to the right state for use. Shouldn't take too long but I've got to leave for some errands today.

DustinGrau, 2024-08-11:
Enum is added, and logic mostly works. I'm in need of 2 new voice prompts for "inner cyclotron LED panel static" and "inner cyclotron LED panel dynamic" for the newest options. The selection via the web UI at least works and I have a toggle option working in the menu: it steps through in the order of static -> dynamic -> disabled (individual).

The only real issue I have is that the 6 lights on the right only come on after changing firing modes which isn't an obvious result of my changing the colors, so something else may have broken that portion.

nomakewan, 2024-08-11:
> The only real issue I have is that the 6 lights on the right only come on after changing firing modes which isn't an obvious result of my changing the colors, so something else may have broken that portion.

Does it happen in all modes, or only Frozen Empire?

DustinGrau, 2024-08-11:
Good question, as I think I was only testing in FE for other changes. I can test further in the morning.

nomakewan, 2024-08-11:
I ask because there was a change in [7a44b04](https://github.com/gpstar81/GPStar-proton-pack/commit/7a44b040cdf4c8fa58c9081e50f19bf942f0e6ea) to address issue #358 . In FE, if the "brass pack" sound is active, the lights on the inner panel should all be turned off. This is normal expected behavior.

DustinGrau, 2024-08-11:
Dammit, I keep forgetting that. So yes, it's that feature that's throwing me off. Also, I may have found a defect there as when the cyclotron is exposed in FE mode, and you're shutting down, the lights do come on for the ramp-down sequence. I believe that's what really threw me off as I thought the missing ramp-up and runtime was a symptom of a problem, turns out I misunderstood the behavior.

That means the only real item remaining here is the new voice prompts which I can't seem to generate correctly (they don't sound right, and I can't seem to record them via my Mac using any built-in audio loopback devices (since you can't download the sound directly).

nomakewan, 2024-08-11:
Good point; the ramp-down is because the brass pack sound is no longer playing, and we don't have a specific sound for the brass pack ramping down. I'll come up with something to fix that though, thank you for pointing it out.

With the sounds, we'll just wait for Michael to make some for us, same as always. Not a big deal.

Aces6469, 2024-08-11:
I could give you guys mine for testing?

nomakewan, 2024-08-11:
Is yours the one from Die Geisterjäger?

Aces6469, 2024-08-11:
With a bit of tinkering to match up the current shutdown length, yeah.

nomakewan, 2024-08-11:
Sure, I'll take a peek at it. I think Github will allow attaching it directly to a reply too.

Aces6469, 2024-08-11:
[Sparkolotron SFX.zip](https://github.com/user-attachments/files/16575524/Sparkolotron.SFX.zip)
Should have the long one that matches the current length, and the original effect just in case you guys want it to make your own versions.

nomakewan, 2024-08-14:
> That means the only real item remaining here is the new voice prompts which I can't seem to generate correctly (they don't sound right, and I can't seem to record them via my Mac using any built-in audio loopback devices (since you can't download the sound directly).

Since we now have the voice prompts and the code seems to be mature, gonna go ahead and say this issue is resolved! Thank you so much Dustin for getting the code part done, and Michael for tossing us some voice prompts on short notice!

closed 2024-08-14 by nomakewan

DustinGrau, 2024-08-14:
Oh, I had managed to find a way to use a "loopback audio device" on my mac that let me record off of Chrome. I still don't know if the voices are "correct" but they work as placeholders for now and let me finish this.