gpstar81/GPStar-proton-pack #346: [Bug]: quick vent pump still not going full time

Toy203, 2024-07-13:
### What actions were attempted?

I extended the Overheat time in eeprom to have a decently long quick venting action (should be around 10 sec once halved). My goal is to be able to trigger a smoke event without the alarm and light slowing down. 

### What actually happened?

I can hear the pump going for a couple of seconds, but then it stops. Fans and lights keep going correctly during the whole sequence of quick vent. 

### What was expected?

The pump should keep going up until the quick vent status ends. 

### Firmware Version

5.3.1

### Mode in Use

Super Hero / Video Game

### Notable Hardware

Wasteland smoke kit

### Homework Completed

- [X] I did my part!

nomakewan, 2024-07-14:
To be clear, do you have "Overheat sync to fan" enabled or disabled? I can see where this bug might be, but it all depends on how you have that setting configured.

As a bonus if you'd like to see what happens when you switch that setting to the opposite of what you have right now and report back how it behaves, that would be very cool.

Toy203, 2024-07-14:
It was disabled and with that settings it does like above. If i enable synch, looks like quick vent works as intended.

nomakewan, 2024-07-15:
Found the problem. This has actually been in the code for a very long time, but the venting sequence when "Overheat sync to fan" is disabled was piggybacking off the "smoke output while firing" timer. This was not intended, but was overlooked.

Will be fixed for 5.3.2.

nomakewan, 2024-07-15:
Found the actual real problem. I made a silly typo. Teehee. I think if you test Power Level 1 with "overheat sync to fan" disabled you'll find it works normally, just not in any other power level.

Fixed for 5.3.2.

Toy203, 2024-07-15:
Pack is currently open for maintenance and updating (buy that new amplifier is good) so i trust your word on this. Keep up the good work guys!

commit 921d6a29 (2024-07-15): Fix quick vent smoke bug

Fixes #346
closed 2024-07-16 by Toy203