gpstar81/GPStar-proton-pack #347: [Feature]: High volume shooting sound

Toy203, 2024-07-13:
### What would you like to add in terms of software changes?

It would be cool to have an option and set the proton stream sound volume separately from the rest. My main thing would be to be able to suprise folks with the firing sound without being deafened by the regular pack droning. Just a random thing I know, I let some kids fire the gun and it's fun but I could imagine hearing an unexpected thunderous sound would be even funnier.  Maybe en eeprom settings to keep it just a few notches over the rest of the sounds. 

### Would this request involve any specific hardware?

Nope

### Homework Completed

- [X] I did my part!

nomakewan, 2024-07-14:
At least for the moment it appears we're out of room for a dedicated setting, but I might try to see if there's a way to "toggle" the system volume to change between volume settings without actually moving what level/option you are on. Something like if you click Intensify it'll change to the next volume setting, while holding it down lets you adjust that setting.

Not going to give any guarantees for this one, but I'll see what can be cooked up for v6.

The-Y33TER, 2025-03-02:
Somewhat similar, why is the boson dart so quiet? All the other modes and alt fires are a decent volume, but the boson dart is barely audible.

nomakewan, 2025-03-02:
I just checked the Boson Dart audio files and they're at maximum volume without clipping, exactly as they came from the 2009 video game. It also didn't sound all that different from the Shock Blast sound. Perhaps because only one part of the Boson Dart sound is actually loud, while the rest trails off, unlike other alt fire modes?

Feel free to inspect the audio file for yourself in your audio editor of choice. If you have Proton Stream Impact Effects enabled, the file in use is `411_boson_dart_fire_impact.wav`. If you have this setting disabled, the file in use is `373_boson_dart_fire.wav`.

Likewise, there is no volume reduction in the code for these files. You can confirm this for yourself here by seeing that the volume variable in use is the normal `i_volume_effects` used for all wand sound effects universally: https://github.com/gpstar81/GPStar-proton-pack/blob/main/source/NeutronaWand/NeutronaWand.ino#L2865

Toy203, 2025-10-03:
I'd say it'll be even better for usability, having the droning fade out after a bit would help during event tremendously. Maybe user could have input on how much time should pass from switch on to fading out too, maybe folks like the constant humming.

closed 2026-03-30 by nomakewan

Toy203, 2026-03-30:
Would it also return momentarily on firing by any chance? (sorry for reopening)
That would be awesome.