gpstar81/GPStar-proton-pack #348: [Bug]: ESP wifi marked as 0000

Toy203, 2024-07-16:
### What actions were attempted?

I've reflashed my ESP32 since wifi  password wasn't working. I flashed all the 4 binaries from scratch (no reset file used).

### What actually happened?

The ESP work as intended and generates the wifi, but the number in the wifi name is 0000
[image]

### What was expected?

It should get a code from the wifi mac from what i recall, it's just weird that it gets 0000 i guess?

### Firmware Version

5.3.1

### Mode in Use

Super Hero / Video Game

### Notable Hardware

ESP32 

### Homework Completed

- [X] I did my part!

joepodell, 2024-07-16:
Just adding a comment, that I can confirm something similar with mine.
My WiFi password, which I never changed, just randomly stopped allowing me to access it.  (Bad password error)
Along with the 3.5.1 upgrade, I manually flashed all 4 binaries (using the reset one).  I then set my own password and then refreshed with the non-reset binary.  My WiFi name also now also ends with 0000.
Since it was finally working again, didn’t give much thought to it, but I guess could be annoying if there are more than one of us at a convention or event with this issue.

Joe Podell
***@***.***

DustinGrau, 2024-07-16:
I've only seen that reported by a few users, but can't understand why it might be happening. In my collection of implemented devices I've used 2 brands from Amazon plus the Frutto-provided ESPs, all with expected behavior. For @Toy203 and @joepodell are you also using the Frutto attenuator kits with the supplied ESP32's? I want to begin tracking if this is specific to those devices or is an ESP issue I've not yet encountered.

DustinGrau, 2024-07-16:
FYI, the reset firmware as it currently exists will be going away officially in 5.3.2 as anyone who has updated since 5.2.2 will have the new override ability to bypass security by holding down the center dial on bootup. And for users with ESP32s as wireless adapters, if they forget their password they have a workaround using a simple pushbutton.

joepodell, 2024-07-16:
Yes, I’m using the one in the kit.  
If you needed me to test on another one… I think I have another ESP32 laying around from before I received your kit.  I could run through the same process, or follow any other instructions you might want me to try.

Joe Podell
***@***.***

DustinGrau, 2024-07-16:
Have either of you had continued issues with disconnects or stability with these devices?

Toy203, 2024-07-16:
I have an amazon ESP32, don't have the full attenuator kit. ESP WROOM 32 if that helps. I was among the first having connection issues with my Android 14 phone, but the pc always connect fine. I had that thing reported before where the correct pw didn't work after last update (def before 3.5, can't remember) so i left it like this until i opened the pack.

DustinGrau, 2024-07-16:
Let me look at this a bit later (not enough time right now), but it might be possible to specify a custom "suffix" or even provide your own SSID. I've not actually tried it, but the plumbing is there for the capability. That might just be the easier option and would mean users have full control over the naming if they have multiple devices. Again, no promises but I'll see what I can do.

joepodell, 2024-07-16:
So far only have had the WiFI authentication issue happen to me once.  (Nice to know now I can use that button push next time.)

Two other issues I’ve noticed since I upgraded to 3.2.1 last weekend, but just haven’t had time to troubleshoot or post tickets about are:
1.  Something isn’t loading correctly on the Overheat Settings page.  Page renders, but I get an error about something not loading correctly and I can’t make or save changes.
2.  Very odd behavior when adding the Frutto Cyclotron light panel.  I’ll go to the page, turn on the setting, then go to update the setting.  Page freezes, and the ESP32 seems to freeze for about 30 seconds.  Have to reload the page, then I can save to EEPROM.  Happens also when I turn the setting off.  Figured it’s just down to the ESP32 thinking very hard about re-organizeing the lights on that strip.

Joe Podell
***@***.***

Toy203, 2024-07-16:
> 1.  Something isn’t loading correctly on the Overheat Settings page.  Page renders, but I get an error about something not loading correctly and I can’t make or save changes.

Got this as well, but it went away once i connected the wand to the pack. Dunno if that's the cause.

DustinGrau, 2024-07-16:
Patience, Grasshoppers. Fixed in 5.3.2. But don't cross the streams--we've got this issue's problem to deal with.

DustinGrau, 2024-07-16:
Hell yeah! Got it worked out over lunch. The 5.3.2 will have the ability to change the default SSID (called "Private Network") on the device preferences screen. You can choose your own custom name and use that as both SSID and the local address (http://YourCustomName.local).

closed 2024-07-16 by DustinGrau