gpstar81/GPStar-proton-pack #351: [Bug]: Frutto Power cell light on Power on not turning on

Toy203, 2024-07-19:
### What actions were attempted?

When i switch the battery on i get two leds from the power cell bar lighting up instead of all 15.

### What actually happened?

I've correctly set up the 15 leds in the eeprom.

### What was expected?

It should flash once from bottom up if i'm not mistaken, like the 13 led option did. Also, would be cool if the inner cyclotron would flash for the lid of power on event.

### Firmware Version

5.3.1

### Mode in Use

Super Hero / Video Game

### Notable Hardware

Frutto 15 led power cell bar

### Homework Completed

- [X] I did my part!

nomakewan, 2024-07-19:
I did make a change to the POST sequence in 5.3.2, though it is a minor one. Still, would like to know if 5.3.2 fixes the issue you are reporting.

Would you be able to test the [5.3.2 pack binary](https://github.com/gpstar81/GPStar-proton-pack/blob/v5.3.2/develop/binaries/pack/ProtonPack.hex) on your pack to see?

Toy203, 2024-07-19:
Will do in 20'

gpstar81, 2024-07-19:
I can confirm that the powercell bootup sequence is still not fixed.

Toy203, 2024-07-19:
Well that was fast lol

gpstar81, 2024-07-19:
Give it a try now.

Toy203, 2024-07-19:
Well it lights up the power cell but i have all white pack lights. Power cell and pucks. Inner cyclotron is completely off, even the control panel leds.

gpstar81, 2024-07-19:
> Well it lights up the power cell but i have all white pack lights. Power cell and pucks. Inner cyclotron is completely off, even the control panel leds.

probably because the eeprom needs to be reset.

Toy203, 2024-07-19:
Thought eeprom would reset automatically now? Apparently i'm having wand connection problem as well, got the top white hat light blinking and cannot enter eeprom menus or turn on the pack from it.

gpstar81, 2024-07-19:
The wand will need flashing. There must have been an eeprom storage setting change which would force a reset.

nomakewan, 2024-07-19:
EEPROM only resets automatically if there has actually been a change to the EEPROM objects. If the EEPROM objects did not change, no reset will occur.

If you only flashed the pack with 5.3.2 that is why your wand is having issues. This was just a test of the POST animation discussed in this issue which only requires flashing the pack to test. If you wanted the entire system to sync you'd need to flash 5.3.2 to the wand as well.

Toy203, 2024-07-19:
Flashed both pack and wand. Cleared both menu (does it need to be done for both?) eeprom but it's all still white. Can confirm eeprom was cleared as it defaulted back to 13 leds etc.

nomakewan, 2024-07-19:
I've pushed a minor update to the previous fix, caught an infinite loop and also a chance of the year switch causing the same bug over again. Also commented out a section that might have been trying to set power cell LEDs to red instead of inner cyclotron panel LEDs.

That said, I don't see anything that would be causing all the LEDs to be white.

Toy203, 2024-07-19:
With this latest update i have my cyclotron panel leds back and working as intended. Still, cycloton pucks, power cell and inner cake led are full white.

nomakewan, 2024-07-19:
Are they white during the POST sequence, or only after?

Are they white when they are turned on when you enter the EEPROM LED Menu?

As I mentioned I don't see anything that would be assigning white to their value (or C_HASLAB which is also white). Very peculiar.

Toy203, 2024-07-19:
They're white both on post and with regular use. Tried changing into videogame modes, still all white (apart for the barrel leds).
Once I enter led menu powercell, inner and outer cyclotron grt back to purple as expected.

nomakewan, 2024-07-19:
Okay, so the calls in the spectral modes are clearly working fine, which means the underlying color calls should be fine.

I'll stare into the void for a while to see if I can figure out what the heck would be generating white otherwise. Thank you for the report.

nomakewan, 2024-07-19:
Okay, figured it out. The new boolean operator I added was actually being used for the saturation value instead of the boolean value, causing saturation to drop to 0. What's odd, then, is that the spectral custom colours worked at all.

New pack binary uploaded, that should do it.

Toy203, 2024-07-19:
Yes, everything looks in ordred now! Nice job. Just a random though, it'd be cool to have a flashing indicator while uploading for the pack as we have for the wand. Power cell maybe?

nomakewan, 2024-07-19:
The wand flashing is actually an "accident". The vent light is pin D13, which in the bootloader for the Mega 2560 is also the bootloader status LED. Likewise this is why the vent light flashes when the wand is first powered on.

The pack has the data pin for the inner cyclotron on D13, which is why nothing happens.

Toy203, 2024-07-19:
I meant for the firmware flashing, but I guess it'll be the same.

closed 2024-07-19 by Toy203

nomakewan, 2024-07-19:
> I meant for the firmware flashing, but I guess it'll be the same.

Correct. Firmware flashing is handled by the Mega 2560's bootloader, which is why the vent light blinks while flashing and also why it turns on when first powered up. Both are because they are indicating the status of the onboard bootloader. It is not intended behavior, and changing that behavior (either to remove the vent light blinking or to blink some other LED on some other pin on the pack) would require writing a custom bootloader. It cannot be done with just a code update.