gpstar81/GPStar-proton-pack #352: [Feature]: Random Smoke Events

Toy203, 2024-07-25:
### What would you like to add in terms of software changes?

During conventions i love to leave the pack running at the booth with the power cable plugged in. I'd like it to be able to trigger a smoke sequence randomly on its own. I know this might strain any battery if left on on normal use, but it'd be very cool functionality to have. Safety wise, the effect could automatically turn off after a certain number of uses or duration i guess, to avoid burning the e-cig if runs out of liquid? Maybe even user selectable.

### Would this request involve any specific hardware?

Any smoke kit will do.

### Homework Completed

- [X] I did my part!

nomakewan, 2024-07-26:
Oooooh. Actually not a bad idea; v6 is looking to change a lot of the random boolean variables in the code into enums or structs where applicable, which also frees up space in the menu. I could forsee having "Demo Light Mode" allowing more than just that, and also allowing "Demo Feature Mode" where the pack can be left alone untouched and will run through a demo sequence based on what year the pack is currently set to.

Kinda like how the Hasbro PKE has a pre-programmed demo mode where it goes through pre-programmed animations in sequence rather than being controlled by EMF or manual user input.

Honestly would love to use that to do things like have the pack replay brief sequences from the films (_Nice shootin', Tex!_) while it's left idle.

Honestly super cool idea and something I wouldn't mind playing with the idea for. Thanks for this, I'll add it to the list!

Toy203, 2024-07-27:
Just as side note though, I burned my kit ecig mistakenly running it when the liquid was finished. It'll take a while for it to finish, didn't do the math on the autonomy of my Wasteland yet.
Damage is trivial as I just replaced the glass ecig vial but better inform the users clearly of the risks.

DustinGrau, 2024-09-11:
Based on the discussion, I would support the idea of a button on the web UI that could let you give off some smoke on demand but not fully vent the pack. Maybe a single button press gives X seconds of smoke (configurable, possibly) so you are in control of when it happens. That way you can time it when someone approaches the pack or gets too close to the prop table (something I've even run into at events).s

Toy203, 2024-09-11:
We basically have that now, although it's missing a quick vent option like we have on the wand, only does it with the full vent sequence. My point was to being able to let it so his thing on its own at random intervals or after a fixed time.

DustinGrau, 2024-09-11:
There's a vent option on the web UI but it's limited to only when the system is powered up via the wand. I do get the idea, but for the reasons mentioned about the smoke juice supply is why I'm leaning more towards the on-demand option. Automatic smoke would require more configs to set duration of the smoke, interval between (random or specific), and overall time to allow it to do the behavior.

nomakewan, 2024-09-11:
That's actually why I was suggesting that perhaps a better solution to fulfill the spirit of this request would be to introduce a demo feature mode.

This would just have the system perform its normal duties (firing, smoke, etc) as it would if you were using it, but instead of the user manually utilizing the system, the system is automatically playing back a clip from the film (like trying to capture Slimer in the Sedgewick ballroom, or the "This chick is toast!" rooftop scene, etc). This way there's be no need for those extra configurations, because the sequences are pre-programmed. It's just a matter of the delay between random events.

DustinGrau, 2025-10-05:
As this is an older request, I'm suggesting we merge this with #386 to consider a broader range of smoke events and consolidate settings when/if that gets implemented.