gpstar81/GPStar-proton-pack #356: Capture Stream

LelandFormhals, 2024-08-01:
### What would you like to add in terms of software changes?

Don’t know if this has been asked already but a capture stream mode would be fantastic, I don’t know exactly what button it could be to activate but it would play the 2009 game wrangle effect sound and maybe even have a ghost squirming sound in the background similar to how the ghost trap is going to have different ghosts. I apologize if this seems like too much, just thought it was a gnarly idea lol.

### Would this request involve any specific hardware?

I don’t think so.

### Homework Completed

- [X] I did my part!

nomakewan, 2024-08-01:
For now the whole "capturing ghosts" thing is a ways off. It is absolutely a neat idea and a few of us have been pondering how to actually go about making it work.

Let's wait for Two In The Box to come out and try again from there. ;)

Toy203, 2024-08-07:
The capture sound events (wrangle, ghost screaming) could be tied to a timer based on a specific power level of the wand, i.e. : if you fire the wand at level 5 for x seconds it triggers the capture sounds. Could be optional on each firing level to have more granular control, like the firing smoke event is.

The-Y33TER, 2025-01-14:
Alternatively, what does the alt fire button do right now when not in a CTS mode while firing? The CTS mode obviously is a mainline movie thing, removing the ability to use 2009 modes, and you'd only use the cap stream when you have 2009 modes on. Meaning, if the alt fire is unused during firing and has no effect, pressing (and holding) both the intensify trigger and alt fire trigger (alt fire AFTER intensify, maybe a safety of 1+ second firing required before the alt fire hold activates the capture stream.) should activate the 2009 SFX for capture stream starting, activating a capture stream until either trigger is released, maybe another safety, making the alt trigger require being released before intensify?

nomakewan, 2025-01-14:
The alt fire button is the alt fire from the 2009 game. So for Proton mode, it fires the Boson Dart.

If you press alt fire while already holding Intensify, it starts the CTS sequence (yes, CTS was a thing in the 2009 game as well).

Two in the Box is out and development has begun on that. Trust me, we haven't forgotten about the capture stream stuff. Just right now, there are a lot of balls in the air so nothing has solidified as to how best to approach such a feature as of yet.

The-Y33TER, 2025-01-23:
I'm aware 09 had CTS, but what does it do when you have CTS off entirely?

nomakewan, 2025-01-23:
There is no way to "disable" CTS. It works as follows:

[u]Video Game Mode - Proton[/u]
Intensify: Proton Stream
Alt Fire: Boson Dart
Intensify + Alt Fire: Cross the Streams

[u]Cross the Streams/Cross the Streams Mix Mode[/u]
Intensify: Proton Stream
Alt Fire: Alt Proton Stream (inverse of power level)
Intensify + Alt Fire: Cross the Streams

The-Y33TER, 2025-01-24:
> There is no way to "disable" CTS. It works as follows:
> 
> __Video Game Mode - Proton__
> Intensify: Proton Stream
> Alt Fire: Boson Dart
> Intensify + Alt Fire: Cross the Streams
> 
> __Cross the Streams/Cross the Streams Mix Mode__
> Intensify: Proton Stream
> Alt Fire: Alt Proton Stream (inverse of power level)
> Intensify + Alt Fire: Cross the Streams

oh damn

DustinGrau, 2025-10-05:
[image]

Coming soon in the 6.1 release will be an option to change the stream mode via the Attenuator dial. While we are still locked into the standard stream options for selection as is available to the wand, this offers another way for a user to change easily. I will propose to the team an idea that a "Capture Stream" be another Proton stream that effectively swaps the use of the Alt-Fire button from CTS to triggering the capture stream. That would free up a button to initiate actions that could potentially work with the forthcoming upgrades to the Haslab Ghost Trap (still being planned for a full hardware/software overhaul). We also have the IR capability in the new GPStar II kits which can trigger the existing ghost traps with their PKE-initiated sound effects, and this would be another way to reuse that alt-fire button to trigger events (eg. such as pressing that button when not firing).