gpstar81/GPStar-proton-pack #358: Disabling The Inner Switch Panel Lighting

ProtonCharging, 2024-08-08:
### What would you like to add in terms of software changes?

The Option of Disabling the Internal Panel Lights without disconnecting any JST connectors. Could take up a spot in the LED menu if there's room, or be a third selection for the panel option on EEPROM LED menu level 2?

### Would this request involve any specific hardware?

Inner Cyclotron LED Kit (Old and Updated)

### Homework Completed

- [X] I did my part!

Toy203, 2024-08-08:
Out of curiosity, why disable it?

nomakewan, 2024-08-08:
Have to agree here. What is the use case?

Implementing it would be trivial for v6 (see #345 ) but I'm curious as to why.

ProtonCharging, 2024-08-08:
I'm using the fairy lights on my proton pack to try and replicate the sparking pack's look. Ends up getting drowned out by the panel lights on the top left, and I don't wanna remove them since it'd end up just being a waste of money for me.

nomakewan, 2024-08-08:
See now that actually makes sense. I'll go ahead and get this into 5.3.3 as a bugfix, and for v6 it can be turned into an option.

nomakewan, 2024-08-09:
Fix has been implemented and will be a part of 5.3.3 as planned.

closed 2024-08-09 by nomakewan

ProtonCharging, 2024-08-26:
Just updated to 5.3.3, are the left two lenses by the switches supposed to still be lit? If so, is it possible to have those two go dark during the brass pack mode too? I wanna get as much spark light as I can. Thanks :)

nomakewan, 2024-08-26:
Those two are the mode indicators for year and vibration.

Yes, I can have them turn off as well.

ProtonCharging, 2024-08-26:
That'd be great, thank you!

commit 7a44b040 (2024-08-09): Update FE brass pack functions

Disables cyclotron switch panel LEDs if brass pack sound is running (resolves #358) and makes it so user-set spectral colors will be respected in spectral custom mode with brass pack sound running (resolves #359).