gpstar81/GPStar-proton-pack #359: Inner Cyclotron Spectral Colors Not working in FE mode

Aces6469, 2024-08-09:
### What actions were attempted?

I edited the inner cyclotron color to white, saved the settings, started up in Frozen Empire mode with the lid off, changed to spectral mode, and the color was the exact same as the normal FE cyclotron color. This only happens in FE mode.

(I've got a link to a google drive upload for the video since it caps at 10mb)
[](https://drive.google.com/file/d/1uLx5YQOsTM_WLrSjuE8H04QORCqCtCPU/view?usp=sharing)

### What actually happened?

I edited the inner cyclotron color, and changed to spectral mode in FE mode. The color was the exact same as the default inner cyclotron color no matter what I did. It worked fine color wise for all other eras.

### What was expected?

I expected the inner cyclotron spectral color to be the one I selected in the LED menu.

### Firmware Version

5.3.2

### Mode in Use

Super Hero / Video Game

### Notable Hardware

Inner Cyclotron LED kit (ring and individual lights, not the pcb), Strip Light around the cyclotron.

### Homework Completed

- [X] I did my part!

Aces6469, 2024-08-09:
https://drive.google.com/file/d/1uLx5YQOsTM_WLrSjuE8H04QORCqCtCPU/view?usp=sharing

nomakewan, 2024-08-09:
This is expected due to the changes introduced to solve #344 . Clearly I misunderstood what you were actually looking to do by enabling "brass pack" for spectral modes.

My question is why you have brass pack sounds in Afterlife. Did you do that yourself, or is that another bug I need to look into?

Aces6469, 2024-08-09:
Sorry about that lol, back then I was meaning to just add 334 (the boot FE boot effect sound) to the spectral modes, not change the way they function. Probably wasn't clear enough on my part, that's on me.

As for the brass sounds in Afterlife mode, the shutdown is expected, since I had to change the afterlife shutdown sound for the brass shutdown to be present in FE mode. As for the startup, that only happens when I startup via the thrower, but I'm not sure if that's a bug or not.

nomakewan, 2024-08-09:
When started from the thrower, Afterlife uses the shorter boot sound rather than the full "switch me on" startup from the ion arm switch. This was because in the film, we only actually see the full startup when switched on from the ion arm switch, whereas the rest of the scenes showing Phoebe firing the pack back up from the wand use the Sedgewick boot sound. This means it's trying to play the "FE" non-brass boot sound, which is file `001.bootup.wav`. Did you change that file to be the brass pack sound?

Aces6469, 2024-08-09:
Yep, thats the one. Which means no bugs to worry about. (yippee!)

nomakewan, 2024-08-09:
All righty then! That being the case, your request is done and it'll be included as part of 5.3.3's release.

commit 7a44b040 (2024-08-09): Update FE brass pack functions

Disables cyclotron switch panel LEDs if brass pack sound is running (resolves #358) and makes it so user-set spectral colors will be respected in spectral custom mode with brass pack sound running (resolves #359).
closed 2024-08-09 by nomakewan