gpstar81/GPStar-proton-pack #360: [Bug]: ESP32 doesn't clear the IP address assigned

Toy203, 2024-08-09:
### What actions were attempted?

I was still tinkering with the wifi settings, it appears to me that even if the checkbox is off the ESP still takes the manually added IP address instead of getting the one assigned from the phone tethering.
In this case, the 11.112
[image]
was manually added in the option but i marked off the option:
[image]

### What actually happened?

...and yet it still takes it. Even if i delete the data in form entirely keeps the same settings, can't make it just take the granted IP from the tethered wifi network

### What was expected?

Got the feeling MAYBE if it took the assigned ip address by default it might finally be reachable from my phone withouth headaches.

### Firmware Version

5.3.2

### Mode in Use

Super Hero / Video Game

### Notable Hardware

ESP32

### Homework Completed

- [X] I did my part!

DustinGrau, 2024-08-09:
This might have more to do with the DHCP lease time. I can't really control the devices that are assigning the IP address, but leaving the toggle OFF should allow the device to pick up whatever has been assigned to it.

I need more explanation though: what is the IP address you manually set? What is the IP range your phone is handing out? Your statement isn't clear as to which the "11.112" is referring to.

Toy203, 2024-08-09:
I assigned the 192.168.11.112 manually in the ESP32. My damn s22 has the infuriating habit of changing the second to last  IP value randomly after some time, so while it was indeed that the correct IP it assigned to the ESP, now it assigns something like 192.168.88.112 or something like that, and it'll keep changing. Such idiotic functionality, I get the benefit but I can't disable it or assign fixed IPs on phone side.

DustinGrau, 2024-08-09:
You may need to just keep overriding with the manual IP then, something that's on the same network and can reach the phone. I honestly wish I knew how these hotspots serve up IP's, as I know even my iPhone gives out some strange IP's.

Toy203, 2024-08-09:
I see, hoped i could somehow wipe the ESP IP data to avoid the second device thing.

DustinGrau, 2024-08-09:
Try to clear all the info, including the SSID and password for the external wifi, and it should reset everything.

DustinGrau, 2024-08-09:
Alternatively, change some character in either of those fields, and it should force the IP info to get fully reset because it will think it's a completely different network to join.

Toy203, 2024-08-09:
Thanks for the suggestion, I'll look into it a bit further then.

closed 2024-08-09 by Toy203

Toy203, 2024-08-13:
It freaking worked!

Toy203, 2024-08-14:
It freaking worked! As a side note, apparently my s22 likes to change tethering ip assignments after each reboot.