gpstar81/GPStar-proton-pack #363: [Feature]: Support pack overheating when a GPStar wand is not attached

DustinGrau, 2024-08-18:
### What would you like to add in terms of software changes?

As we now have the ability to connect a stock Haslab wand to a GPStar pack controller and react to Proton firing events, we need a way to control and initiate an overheat action based on the firing by a stock wand. We seemed to have forgotten that the timing values are only stored on a GPStar wand controller and not the pack, making these values and the triggering of the event impossible under the current arrangement.

### Would this request involve any specific hardware?

No new hardware, but this would be applicable to cases where a user has installed the Frutto Current Sensor for the wand and expects to use a stock Haslab (Spengler or 1984) Wand with their pack.

### Homework Completed

- [X] I did my part!

DustinGrau, 2024-08-18:
Assigned to myself for now, as I'm working on making sure the preferences are at least capable of being sent back and forth to the Attenuator (web UI) when a GPStar wand is not connected.

nomakewan, 2024-08-22:
I had mentioned it in a chat message but I forgot to actually put it here for posterity.

Instead of using timers, we should use the power draw from the wand. The stock Hasbro wand itself has an overheat built right in which causes the wand to shut down completely while firing. So from the pack's perspective we should see the wand in its firing state, and then suddenly shut off without first returning to idle. We should be able to use that to mimic the code that runs when `W_OVERHEATING` is received by the pack's serial buffer.

Sure we could also put a rough timer in place since the stock wand always starts its overheat sequence at around 17 seconds of continuous firing too just to handle edge cases of a user flipping down the vent switch mid-firing or something, but yeah.

nomakewan, 2025-04-21:
This is now available in 5.4.Bugfix and will be included in 5.4.4 releasing this week.