gpstar81/GPStar-proton-pack #364: [Feature]: Stock wand power triggering fine-tuning

DustinGrau, 2024-08-18:
### What would you like to add in terms of software changes?

We should offer an ability to fine-tune the power values returned by a stock wand so the user can adjust to their needs. This is more important for those who may "soft mod" their wand, replacing some stock components but not upgrading to a full GPStar controller which offers serial communications. This would let users get a power reading from their wand while performing an action, and use that to set a threshold for an event: namely when to consider the wand as powered on (and thus turns on the pack), and when to consider a proton stream is being fired.

### Would this request involve any specific hardware?

Only requirement is the Frutto Current Sensor with a stock Haslab wand

### Homework Completed

- [X] I did my part!

DustinGrau, 2024-08-18:
First working on the ability to send/display the latest power reading from the wand on the web UI as this will be the only means of making these adjustments.

DustinGrau, 2024-08-21:
Data can be sent and displayed on the web UI now. Just need to work on a way to send back some custom values and use those as the thresholds for "power on" and state-change triggers.

nomakewan, 2025-01-22:
Converted issue to bug report as we have had reports from several GPStar Lite users regarding undesirable operation at certain extremes (Power Level 1 operation, rapid power level changes). As these are caused by the use of absolute value thresholds in software, the fix will not be calibration of said thresholds but rather a transition to a new paradigm.

I have collected several datasets using a stock wand and theorize that the best solution here is to use a 4-value-wide sliding window to determine magnitude-of-change of the input current in order to detect wand states. This is because while there is quite a bit of variation with absolute value thresholds in wands, the current spikes due to change in vibration motor state on unmodified wands is quite distinct. We read one value every 10 ms, so this detection should only take 40ms to trigger each state change. According to my data, 4 values is likewise sufficient to distinguish the inrush current of the vibration motor from other state changes.

This fix is targeted for 5.4.4 and will be worked on as soon as 5.4.3 is public.

nomakewan, 2025-04-21:
Going to close this issue now as the 5.4.Bugfix branch now has the new tuning in place which seems to be working quite well indeed, and 5.4.4 with these changes should be out within the week. Hooray!

closed 2025-04-21 by nomakewan