gpstar81/GPStar-proton-pack #367: [Feature]: Ice effects on the cake

Toy203, 2024-08-24:
### What would you like to add in terms of software changes?

This might be a stretch but figured to throw it in the ring. How about in the frozen state of FE (when Intensify is mashed too fast), instead of turning the cake ring light off, the light slows down and then it just could be going all led on with a light blue until it gets back on working mode to simulate the ice around it?
might look cool, might look wonky dunno. 

### Would this request involve any specific hardware?

Just the cake ring led. 

### Homework Completed

- [X] I did my part!

DustinGrau, 2024-09-10:
Interesting. So basically once the cake "spins down" it just goes blue? The movie effect was to cause the pack to power down entirely so there was no movement (sparking) or system lights at all.

Toy203, 2024-09-11:
Yes it was just a passing though, and it could not work at all to be honest for folks who has a transparent lid in the cake. But i thought at the look of the entire pack shut down and the cake just glowing light blue for few sec could be cool

DustinGrau, 2024-09-11:
Implementation of a switch to have a "Ice Over" lighting option is easy, but the animation is harder. We can leave this request open, and consider it for a possible later release. We've been talking about refactoring the cyclotron code which is grown quite complex with the animations currently in use. It doesn't leave us much room to alter the sequence as it stands today.

Toy203, 2024-09-11:
Well I like it for sure! XD

![Uploading
[image]
 20240911_130936.jpg…]()

DustinGrau, 2024-09-11:
Yeah, I was thinking if we could do some kind of low flickering effect it would look good. Steady-on isn't likely to give the same effect, and there's lots of animations we could potentially do if we can just break free of the current set patterns.