gpstar81/GPStar-proton-pack #368: Halloween Spectral Swap

Aces6469, 2024-08-25:
### What would you like to add in terms of software changes?

Kinda surprised I haven't seen anyone ask it yet, but possibly an LED setting that changes the colors of the christmas mode to purple and orange for Halloween? 

### Would this request involve any specific hardware?

Frutto Powercell, 5 or 9 LED PCBs.

### Homework Completed

- [X] I did my part!

DustinGrau, 2024-08-25:
This does actually exist, it's just not exposed as an option. I started to add some other color combos as I figured people may want sports team colors or other holidays, namely Halloween. So the ability isn't far off, just not made into a user preference.

DustinGrau, 2024-08-25:
Look at https://github.com/gpstar81/GPStar-proton-pack/blob/main/source/ProtonPack/Colours.h
C_ORANGEPURPLE was intended to do just this thing

nomakewan, 2024-08-25:
Yep, it's been in the code since literally forever. We just have to figure out some way to select _which_ holiday and it'll be good to go.

If you have any suggestions on how to switch between Christmas and Halloween for the Holiday mode, we're all ears!

Aces6469, 2024-08-25:
Could use the wing button at the end of the thrower? Last I recall it does nothing in the spectral modes with all toggles up.

nomakewan, 2024-08-25:
And so the solution presents itself instantly. Gonna go ahead and assign this to myself and get it ready for 5.3.4.

nomakewan, 2024-09-07:
Holiday modes for both Christmas and Halloween are now fully implemented and will be in the 5.3.4 release. Thank you for your suggestions!

commit e362d690 (2024-08-25): Implement Halloween holiday mode

When wand is in Holiday firing mode, user can now press the barrel wing button to switch between Halloween and Christmas modes, which are fully supported on the wand, pack, and attenuator. Resolves #368
commit e4c9113f (2024-09-07): Fix 28-segment bargraph overheat, add holiday sfx

This fixes the top two segments of the Frutto 28-segment bargraph becoming stuck if the "blink bargraph while overheating" setting is disabled which resolves #376 and both adds some sound effects for switching between holiday stream modes and updates colour transitions for those modes to improve perceptibility which finally resolves #368
commit 97e0c2a3 (2024-10-08): V5.4.0 (#381)

* Code cleanup

This cleans up a lot of unused variables from the codebase, corrects the documentation on the inner cyclotron LED ring, and fixes a bug where the wand would not play the correct voice clip when changing between static and dynamic inner cyclotron LED panel modes.

* Implement Halloween holiday mode

When wand is in Holiday firing mode, user can now press the barrel wing button to switch between Halloween and Christmas modes, which are fully supported on the wand, pack, and attenuator. Resolves #368

* Fix barrel tip fade effect

This fixes the speed of the barrel  [... cut]
closed 2024-09-07 by nomakewan