gpstar81/GPStar-proton-pack #369: [Feature]: Add POST animation for Inner Cyclotron

nomakewan, 2024-08-25:
### What would you like to add in terms of software changes?

Add a simple animation to the inner cyclotron area in case the lid is off when the user turns their battery on.

### Would this request involve any specific hardware?

Any inner cyclotron switch lights or cyclotron cake ring LEDs.

### Homework Completed

- [X] I did my part!

nomakewan, 2024-09-07:
This is now partially implemented in commit bcb1e7a.

I would have just flickered the cake but commanding all of the possible 36 LEDs to red at full brightness at once would've potentially drawn 0.7A of current so probably not a great idea for a silly bootup sanity check. Will have to get a little more creative.

Toy203, 2024-09-08:
Yeah I can see the issue. Maybe just one led per "quadrant"?

nomakewan, 2024-09-08:
I did think of doing that, but that gives us a slight problem because unlike the outer cyclotron LEDs, we have no concept of "center" or "alignment" on the inner cake ring. It was never intended to operate like the '84 outer ring, only as a smaller Afterlife inner ring.

So right now I'm looking at the actual POST control loop and seeing if I can just run a quick ring around the LEDs outside of the existing controls but inside of the existing timing.

There are three timing loops in the current POST sequence, and the first two vary in length depending on how many LEDs you have in the powercell. If you have 13 LEDs, the first two take 780ms to complete. If you have 15 LEDs, they take 900ms to complete. The final loop is independent of the powercell and takes 1300ms to complete.

So what I'm thinking, just to get something done, is to write a little code in the third timing loop that does a lap of the ring once every 30ms or so. I'm just trying to come up with a way to do it that isn't super overcomplicated and doesn't rely on yet another RAM-hungry timer. I'm thinking of using modulo against a counter variable, but we'll see. I'll be a little busy the next day or two so it might take me a bit to get it done, but I'll get something done.