gpstar81/GPStar-proton-pack #372: [Bug]: Non-addressable barrel LED not flashing during entire firing sequence on Single Shot

nomakewan, 2024-08-26:
### What actions were attempted?

Fire the Single Shot wrist blaster.

### What actually happened?

The BARREL LED flashed once at the end of the firing sequence.

### What was expected?

The BARREL LED should flash through the entire firing sequence.

### Firmware Version

5.3.3

### Mode in Use

Any / Both Modes

### Notable Hardware

Single Shot Wrist Blaster with non-addressable barrel LED.

### Homework Completed

- [X] I did my part!

DustinGrau, 2024-08-26:
Arguably this isn't a bug, the idea was to always use addressable LEDs. Michael has a new version on the way that will be usable for both the single lens cyclotron and the barrel, allowing us to use colors for the effect. I thought I had left the W-LED port active but there may have been a timer removed that manages that.

nomakewan, 2024-08-26:
You did leave it active; but your firing uses a lot of the semi-automatic code. In the semi-automatic firing modes, the barrel tip LED only illuminates when the pulse finishes traveling up the barrel. But in this case, there is no barrel, so the code controlling the BARREL LED port needs to be modified to flash throughout the pulse sequence rather than only flashing once at the very end.

commit 18778640 (2024-09-07): Static barrel LED support for blaster

This change allows the BARREL LED pin to be pulsed when firing the single shot blaster, which resolves #372
closed 2024-09-07 by nomakewan