gpstar81/GPStar-proton-pack #374: [Bug]: FE overheat cut sounds to the pack

Toy203, 2024-09-04:
### What actions were attempted?

Triggering the FE overheat sequence cuts any sound bit from playing. I can hear the first warning beep (from the shut down sequence, not the initial alert) but the sounds then don't play. 

### What actually happened?

Lights perform the shutdown sequence correctly, the pack restart the functions but no audio is played. Green led is static on the gp audio board. 

### What was expected?

The only way to get sound back is to disconnect the battery. 

### Firmware Version

5.3.3

### Mode in Use

Super Hero / Video Game

### Notable Hardware

Gpstar audio board. 

### Homework Completed

- [X] I did my part!

nomakewan, 2024-09-04:
Could you test 5.3.4 beta to see if the bug is still there? Binaries are up to date: https://github.com/gpstar81/GPStar-proton-pack/tree/v5.3.4/develop/binaries

I actually already fixed a few things related to Frozen Empire sound playback in the pack code, so I need to know if this bug was in fact already fixed while I was in there. Note that you'll need to update the SD cards as well (though miracle of miracles you won't need to download 840MB now, only 300MB).

Toy203, 2024-09-05:
Can't replicate on 5.3.4b

closed 2024-09-05 by Toy203

nomakewan, 2024-09-05:
Thank you so much for testing! Good to know it got fixed while we were already in there picking it apart.