gpstar81/GPStar-proton-pack #379: SFX Care Package

ProtonCharging, 2024-09-11:
### What would you like to add in terms of software changes?

Not sure which one of the issue templates this should go in, so sorry if you wanted it somewhere else lol. I Decided to throw together a quick SFX care package for you guys. It's got a few different audio files inside, but the most important in my opinion is the brass pack files. I included a more accurate brass shutdown since I know separating it into it's own mode is in the cards. Should just need a new animation synced to it, since I've ensured it works with the hardware. Don't forget to check the Readme I included, it has a little bit more info that ought to help explain more. Let me know if you've got any questions :)

(I had to split them up since github limits uploads to 25mb each)

[SFX Care Package.zip](https://github.com/user-attachments/files/16970996/SFX.Care.Package.zip)

[SFX Care Package Pt 2.zip](https://github.com/user-attachments/files/16970997/SFX.Care.Package.Pt.2.zip)

### Would this request involve any specific hardware?

nope

### Homework Completed

- [X] I did my part!

nomakewan, 2024-09-12:
Hey there!

First, thank you for these. At the very least I'm going to go ahead and add the Boson Dart w/ explosion and code it so that it'll play if the user has the "proton stream impact effects" toggle enabled. Figure that's a good compromise. ;)

The Meson Collider Open sound being the way it is was actually done because we didn't have any mechanical componentry that would associate with the sound, whereas the Stasis Open was simpler (like the Slime Open) and so sounded less intrusive. But since you're now at least the second person to request the actual TVG sound for this, I'll go ahead and put it in for 5.3.4.

With the brass pack, I know where you got these files from, and generally speaking they're intended for monophonic audio use (thus why the powercell idle loop is layered into the brass pack startup and idle, for example). In fact, in the film at no point do you hear the powercell idle for the brass pack sequence, so I'm hesitant to use any of these. That being said, if you can isolate the metallic rattling sound that's been layered into the pack idle (something I asked Die Geisterjäger but got no response on) I'd love to use that instead of looping the startup.

The firing sounds on the other hand are pretty darn good, so I'll probably use these. It'll be even better once 6.0.0 makes the brass pack its own thing rather than Frozen-Empire-without-Lid-Cross-the-Streams. Thanks!

The "1984 idle with afterlife cycling effects" is in fact already in as the Frozen Empire idle loop. ;)

I like your take on the wand mash error a lot. Definitely going to use it for 5.3.4.

And finally, with the Afterlife startup and idle loop, that's actually a project I've been working on in the background myself. I think I have a pretty good option there but I'll probably be opening a poll in the FB group to see which versions people like the best and whichever wins the blind poll will end up in 5.3.4 and beyond. I actually just got my pack speakers in the mail today, so I should be able to test all of these on real hardware this week.

Thank you very much for your submission! I'm going to "close as completed" since it's not really an issue per se, but the discussion portion will remain open, so feel free to make comments and/or suggestions relating to these files!

closed 2024-09-12 by nomakewan

ProtonCharging, 2024-09-12:
Glad to hear some of my stuff is getting put to use. If it sways you at all though on the idle, we did hear the power cell beep from the brass pack. In the basement scene when phoebe confronted melody. But we also heard the 84 startup in that scene, and a different startup later on during the actual sparking shot, so do with that as you will lol.

nomakewan, 2024-09-12:
To be fair the scene with Melody is completely inconsistent. We don't hear any brass effects whatsoever (even though with the camera that close we should be able to hear the inner cyclotron rattling), and supposedly the pack and wand are on yet the only effect you hear as she descends the stairs is the powercell. Then as Phoebe reaches the bottom of the stairs, by magic the wand is off again so she can turn it back on again for dramatic effect. So no, I'm going to disregard the stairwell scene in favor of the actual brass pack scene.

And besides, again, the problem is these sounds were intended for a monophonix audio board, and we're polyphonic. We already have the powercell beep as a separate effect. So while I'd love to at least use this idle, I'd need a version that at minimum has no powercell, and at best is just the nice clean metallic rattle sound so we can just have it mix into the normal FE idle.

Aces6469, 2024-09-13:
Not sure if this helps or not, but I managed to reduce the power cell sound down to where it's barely even audible anymore. Still has the background afterlife idle though, so not sure if you guys want it or not.

nomakewan, 2024-09-13:
I'll give it a listen. If it loops okay and gets "close enough" it still might be better than what we've got.

Aces6469, 2024-09-13:
Apologies on the wait, had some stuff to do. Here ya go. If you want a video I can dm it to you on FB too.
Went ahead and ran the startup, and the shutdown from Geisterjaeger through the same filters so they'd match up.

[Brass (no power cell).zip](https://github.com/user-attachments/files/16998648/Brass.no.power.cell.zip)