gpstar81/GPStar-proton-pack #380: Mechanical Cyclotron Support

Aces6469, 2024-09-11:
### What would you like to add in terms of software changes?

Following up from the facebook group, would it be possible to have an option where the booster smoke port works as a 5v power output instead of just smoke? Specifically when the pack is in startup, idle, firing, and shutdown, as I'd be using it to trigger a relay for my 12v cyclotron motor to spin while the pins are hot.

### Would this request involve any specific hardware?

No

### Homework Completed

- [X] I did my part!

DustinGrau, 2024-09-11:
What is the hardware (motor) you plan to use?

Aces6469, 2024-09-12:
https://www.amazon.com/gp/product/B072N84V8S/ref=ox_sc_saved_image_6?smid=A18IHNL4DD28Y&psc=1

Planning on using this one.

nomakewan, 2024-09-12:
That motor draws 25 _amps_. That's beyond the reach of any talentcell. How are you planning on powering it?

Aces6469, 2024-09-12:
Crap, wrong motor.

Aces6469, 2024-09-12:
Meant this one, sorry. They're right next to each other in my list lol.
https://www.amazon.com/gp/product/B0B7BTY64R/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1

nomakewan, 2024-09-12:
There we go, that's a much more acceptable 300 mA. Still high but not unreasonably so.

As long as you're triggering it through a relay then it should work okay. The question now is how to implement this in the code. Or, perhaps more accurately, where the heck to put the configuration option for such a thing.

Do you have a wifi module or Attenuator? The easiest solution here would be to make this a beta option for now that is only accessible via the web interface and leave it out of the wand menus.

Aces6469, 2024-09-12:
I don't have either yet, but I've been planning on getting a module soon, so that works out for me.

nomakewan, 2024-09-12:
Okay. So to confirm, you need the motor to be on whenever the pack is on and not in an error state (ribbon cable off, overheating, mash error), correct?

So the only times the motor shuts down are if the pack is turned off, if the pack overheats (if overheating is enabled), if the pack's ribbon cable is removed (if ribbon cable switch is enabled), or if the mash error is triggered. Correct?

Aces6469, 2024-09-12:
Yep, hit the nail right on the head.

gpstar81, 2024-09-12:
Just a suggestion, but perhaps we can re-purpose the green vibration motor connection for the usage of this in a menu system. Switching between a regular vibration motor or this cyclotron setup instead.

nomakewan, 2024-09-12:
> Just a suggestion, but perhaps we can re-purpose the M1 green vibration motor connection for the usage of this in a menu system. Switching between a regular vibration motor or this cyclotron setup instead.

That...would actually be the easiest solution, since it would mean we wouldn't have to guard off all the booster smoke calls. And if we use digitalWrite() it should override the PWM frequency stuff without any issues (since I already do that on the wand for lighting calls).

Great call, I like that a lot better. Heck I should be able to get that done for 5.3.4.

Aces6469, 2024-09-12:
Welp, there goes my spare time until halloween lol

Aces6469, 2024-09-12:
50/50 on the lid stuff considering some people might take out their interior to do an AL style mechanism for cyclotron lighting. Could go for a 3 part option? One with lid detection, one without, and just the normal settings stuff?

nomakewan, 2024-09-12:
Lid detection generally will be an option (just like use of the ribbon cable switch is now) in 6.0.0. So once that option is available, disabling lid detection will allow the motor to run all the time. I agree that for power saving reasons, the default should be that if the lid detection is enabled and the lid is installed, the motor should not be running.

nomakewan, 2024-09-13:
All righty folks, the support has been coded in and will be included with 5.3.4.

In the end, I decided not to use lid detection but rather use the vibration switch. If you flip the switch off, it'll disable the motor. If you switch it on, the motor operation will be determined by the pack code.

I did this because I realized that this can be used not only to motorize an inner cyclotron cake for brass pack purposes, but can also be used for the folks who wanted to imitate the Afterlife-style bulb-on-a-motorized-arm outer cyclotron in their custom packs.

Have fun!

closed 2024-09-13 by nomakewan

Toy203, 2024-09-13:
Trying to resist the urge of having a rotating and lighting up cake.

Aces6469, 2024-09-13:
Hey if my setup works, keep an eye out on the facebook. I'll most likely post how I did it if I remember.

commit e994681d (2024-09-13): Vibration refactor, cyclotron motor

This completely refactors how vibration works in all devices by switching from discrete bools to enums, and adds the ability to use a motorized inner or outer cyclotron relay in place of the vibration motor on the pack which resolves #380