gpstar81/GPStar-proton-pack #384: [Feature]: Add diagram for webUI graphical pack status screen to WIRELESS_OPERATION.md

nomakewan, 2024-09-14:
### What would you like to add in terms of software changes?

As a revision to WIRELESS_OPERATION.md it may be good to have a version of the graphical interface image that has arrows pointing to each of the overlays in the graphical interface and explaining what exactly they mean for end users. I have to admit that even for me I had to walk the Attenuator code in order to figure out that the green overlay on the Booster Tube meant "system voltage reads higher than 4.2V".

### Would this request involve any specific hardware?

ESP32 Wireless Module/Attenuator

### Homework Completed

- [X] I did my part!

DustinGrau, 2024-09-14:
That's a good point there's probably some conflicting or cryptic meaning behind some of the values. Some of the behavior was meant to be observed as you're using the pack, but per your example I can see where you wouldn't know what certain values or color schemes indicate.

DustinGrau, 2024-09-15:
A diagram with callouts may take a bit more work, but I've got some text to explain what's shown on the CRT screen since that doesn't have clear labels like the text-based display. That should suffice for now.

closed 2024-09-15 by DustinGrau