gpstar81/GPStar-proton-pack #386: [Feature]: Smoke events on ribbon detach

Toy203, 2024-09-17:
### What would you like to add in terms of software changes?

Would be super cool to have the option of getting smoke coming out once you detach the ribbon cable. The alarm we have in place sounds lovely ominous, the smoke and nfilter strobo would look even more scary. 

### Would this request involve any specific hardware?

Any smoke kit

### Homework Completed

- [X] I did my part!

nomakewan, 2024-09-17:
I'm tentatively slapping the 6.0.0 tag onto this, but I'm not sure about this one from an in-universe perspective. If it weren't for that pesky scene in Frozen Empire, maybe we could consider something like what exactly the ribbon cable does and why removing it would potentially cause smoke from the N-Filter.

But in Frozen Empire we clearly see that removing the ribbon cable simply makes the Proton Pack non-operational but safely, and that plugging it back in while the pack is otherwise powered on starts the pack per normal. This would suggest that it's just a control loop (not used for particle high-voltage electrical flow) and that without the control loop the rest of the cyclotron simply does not operate.

Toy203, 2024-09-17:
I agree, my idea was mostly based on the original haslab feature. In 84 mode sounds like the storage unit shutdown alarm so I thought of that scene. I also had fun at the booth pretending something was suddenly going very wrong with the pack while kids were around.

joepodell, 2024-09-17:
Maybe not STRICTLY canon, but I like the idea if it’s detached while the thing is running.  FE just shows us the situation from an off state.
From an on state, I bet the pack would ’SCRAM’ quickly if its logic was disconnected.  All that might create some excess… heat.  :)

nomakewan, 2024-09-17:
Ooooh. Okay, I like the idea of splitting it between startup and operation. Having startup be nothing (essentially just nothing happens until you plug it back in), but if the cyclotron is actually active and you yank the ribbon cable, now nothing is actually controlling the cyclotron.

Okay, that makes sense. Still definitely a target for v6.0.0 but should be fun. :)

DustinGrau, 2025-10-05:
For sake of scoping work I'm going to suggest potentially combining #388 with this request as part of any ribbon cable functionality changes.