gpstar81/GPStar-proton-pack #387: [Bug]: Attenuator (esp32) vibration stops working

Shoop64, 2024-09-17:
### What actions were attempted?

Once a tone is played on the buzzer, the vibration seems to stop working.  

I made several changes to the code and finally got to the point where I can get both the buzzer to work and vibration to work, although I'm not sure it matches the intended functionality.  

The changes I made were using ledcAttach for the buzzer pin and a simple pinmode output for the vibration pin.  Then you can use ledcWriteTone(BUZZER_PIN, i_freq) to play a tone(or a freq to 0 to stop it) and then vibration I just use digitalWrite with HIGH and LOW.

### What actually happened?

I've been trying to build the circuit on a breadboard and was confused about why I couldn't seem to get the vibration to work.  Eventually I worked out that it will no longer work after a tone has been played.  I'm not sure why.

### What was expected?

The vibration should continue to function independent of the buzzer function.

### Firmware Version

20240823180629 build date

### Mode in Use

Any / Both Modes

### Notable Hardware

I think I'm mainly using what it in the BOM for the diy attenuator.  

### Homework Completed

- [X] I did my part!

nomakewan, 2024-09-17:
Are you on an ESP32 or Arduino Nano?

Shoop64, 2024-09-17:
ESP32

nomakewan, 2024-09-17:
Well what a coincidence, I'm neck-deep in ESP32 code right now trying to beef up the serial data sync. I'll go ahead and add this to my plate and see if I can pinpoint the cause so we can add it to the list of fixes for 5.3.4.

nomakewan, 2024-09-17:
Okay, found the problem. The problem is that the tone library actually uses ledc internally, and so its internal declarations were un-setting the ledc definition for the vibration motor that had been initialized in setup.

Now that I know how tone works I'll go ahead and get this fixed so that both the buzzer and vibration motor will work as intended. :) Thank you for the report!

nomakewan, 2024-09-18:
Okay, we have a push that should resolve it. The correct way to resolve this issue would be to wait for the next stable release of ESP32 (should be 3.1.0), then use the proper channel setting for Tone. However, considering we have no clue when that will be, I've included a workaround that _should_ work in the meantime.

You can actually test it yourself; grab the Attenuator code from the main branch (5.3.3, which is build date `20240823180629`), then change `ledcAttach(VIBRATION_PIN, 5000, 8);` on line 158 of Attenuator.ino to `ledcAttachChannel(VIBRATION_PIN, 5000, 8, 5);`. Then replace `pinMode(BUZZER_PIN, OUTPUT);` on line 155 with `ledcAttachChannel(BUZZER_PIN, 440, 10, 0);`.

That should allow both the buzzer and vibration motor to work normally (since we are now initializing ledc on the buzzer using channel 0, and ledc on the vibration motor using channel 5).

Shoop64, 2024-09-18:
Excellent, that makes sense to me.  Thanks!

Shoop64, 2024-09-18:
I tested out the change and I lose the buzzer doing the change on line 155.  Leaving it as pinMode(BUZZER_PIN, OUTPUT); but changing the vibration pin to ledcAttachChannel got both working.

nomakewan, 2024-09-18:
Interesting; I wonder why enforcing channel 0 on the buzzer caused a failure. Perhaps some undocumented feature of the auto-channel? Argh.

Either way, will go ahead and put the pinMode back where it was (with a note that once ESP32 3.1.0 is out we can fix this one far more reliably). Thank you for testing!

closed 2024-10-08 by DustinGrau

Shoop64, 2024-11-01:
This bug appears to be back in the latest code.  It took me a while to get vscode/platformio set up so I could build and test against my attenuator, but it appears back.  Changing to code to use digitalWrite to set high or low and declaring it with pinmode output instead got it working.

DustinGrau, 2024-11-01:
So was this on the 5.4 or 6.0 code? I saw the note about the ESP library change so I'm curious/confused as to whether that's the culprit, and to which release this fix belongs?

Shoop64, 2024-11-01:
I'm working off the 5.4.2 source.

nomakewan, 2024-11-01:
Try the 6.0.Preview source instead and see if that resolves the issue. It uses a different ESP32 core and a newer LEDC command.

Shoop64, 2024-11-01:
I'll give that a shot and get back to you.

Shoop64, 2024-11-01:
Yeah, appears fixed in the 6.0 preview.

nomakewan, 2024-11-01:
Excellent. Then for 5.4.3 we'll make sure to backport those changes. Guess it was a bug in the 2.0.17 core, where despite having an explicit command to override the LEDC channel, it does it anyway.

Thanks for testing!

Shoop64, 2024-11-01:
Yeah, glad to help.

DustinGrau, 2024-11-01:
Just to catch up, are we saying the latest ESP library provides a fix automatically, but we need a more explicit change for the 5.4 code?

nomakewan, 2024-11-01:
The fix for this is to use the 3.0 core with the single `ledcAttachChannel(VIBRATION_PIN, 5000, 8, 5);` command to setup ledc for the vibration pin.

commit 285e9b46 (2024-09-18): Attenuator, serial data fixes

Improved Attenuator pack sync speed and reliability, reduced variable count on Nano, fixed incorrect PWM port on wand vibration motor, added new serial com definitions, and specified channel for ESP32 vibration motor so buzzer does not overlap which resolves #387