gpstar81/GPStar-proton-pack #388: Ribbon Cable Trigger on the attenuator/ESP32 site

Aces6469, 2024-10-07:
### What would you like to add in terms of software changes?

Surprised it hasn't been considered yet or I'm just blind. Could we get a trigger to have the ribbon cable open or closed on the site? I'm one of the people that have a static cable clamp so it'd be nice to have access to the feature.

### Would this request involve any specific hardware?

ESP32/Attenuator

### Homework Completed

- [X] I did my part!

DustinGrau, 2024-10-28:
Just so I'm clear, are you asking for a button on the web UI to manually trigger the ribbon cable alarm effects? If so, there's plans to open up the full APIs so we can trigger any event on the pack or wand also via the ESP32 device, but that's a ways off. That feature needs to be implemented before we can open up other actions. The other one requested is triggering stream mode changes via the web UI, but again requires this same underlying feature.

Aces6469, 2024-10-29:
Yep, hit the nail on the head.