gpstar81/GPStar-proton-pack #390: [Bug]: Weird audio snippet on Battery On

Toy203, 2024-10-08:
### What actions were attempted?

Switching the battery on plays a sound bit before the regular Power On sound

### What actually happened?

https://youtu.be/S9-L27-sWBw

### What was expected?

No weird audio bits lol

### Firmware Version

5.4.0 - 1.03

### Mode in Use

Super Hero / Video Game

### Notable Hardware

Nope

### Homework Completed

- [X] I did my part!

gpstar81, 2024-10-08:
It is because your wand has not been flashed yet to the newest firmware.

Toy203, 2024-10-08:
Flashed all three and cleared  eeprom. Didn't try to plug it off and on to the pack though, let me see.

Toy203, 2024-10-08:
Flashed both pack and wand again, no luck. Might try with a "clean" sd card with just regular sounds to see if that change anything.

DustinGrau, 2024-10-08:
Well that's odd. I had on occasion heard the startup fire after battery-on, but not 100% of the time. The trailing audio on that is VERY odd though. Make sure you're fully updated with all sound effects.

nomakewan, 2024-10-08:
> Well that's odd. I had on occasion heard the startup fire after battery-on, but not 100% of the time. The trailing audio on that is VERY odd though. Make sure you're fully updated with all sound effects.

The trailing audio is their own custom POST sound effect. ;)

Toy203, 2024-10-08:
Correct, POST is edited. Tried with clean SD and just reg sounds, same behaviour.

nomakewan, 2024-10-08:
Problem identified and resolved in GPStar Audio firmware. Thank you so much for this report!

nomakewan, 2024-10-08:
Confirmed fixed in 92691f5.

closed 2024-10-08 by nomakewan