gpstar81/GPStar-proton-pack #391: [Bug]: Yellow Power Cell lights for Slime Blower

JMills75, 2024-10-08:
### What actions were attempted?

Turn on the pack, VG modes enabled.
Confirm VG colors are enabled for the power cell.
Switch to Slime blower mode

### What actually happened?

Power cell lights are yellow instead of green/pink (depending on the year)

### What was expected?

Power cell lights will be green (Pink for 89)

### Firmware Version

v4.0

### Mode in Use

Super Hero / Video Game

### Notable Hardware

15 led power cell

### Homework Completed

- [X] I did my part!

JMills75, 2024-10-08:
Found out it was intentional.  Sorry.

JMills75, 2024-10-08:
Planned change, closing bug

Toy203, 2024-10-08:
I thought it was actually intentional to simulate the slime blower lights? As it makes no sense power Cell get filled with slime? I definitely remember a conversation on that at least.

nomakewan, 2024-10-08:
> I thought it was actually intentional to simulate the slime blower lights? As it makes no sense power Cell get filled with slime? I definitely remember a conversation on that at least.

It was intentional (by me). But it was removed from the code in 5.4.0 precisely because people would think it was a bug rather than a feature. However, after removing it, the pack binary didn't get recompiled.

So for now it will stay a "feature" and based on feedback it'll either stay or be removed for 5.4.1 (assuming any more bugs get found).

Toy203, 2024-10-08:
Well sign one vote for yellow power Cell then.

JMills75, 2024-10-09:
Make it two