gpstar81/GPStar-proton-pack #401: [Feature]: Add a max volume limit option.

rchoiselat, 2024-10-25:
### What would you like to add in terms of software changes?

The system allows for saving a default startup volume as a setting in the attenuator interface. My request is to add a Max volume override that would prevent the system from being turned up via the encoder past a certain level. This would avoid the need in some cases to have an attenuator on the amp itself to lower its overall volume and avoid blowing speakers if someone turns the volume encoder with gusto, not knowing its sensitivity. 

### Would this request involve any specific hardware?

No, this should not.

### Homework Completed

- [X] I did my part!

nomakewan, 2025-05-07:
After reviewing this issue further I do not think it is actually a good feature request. The maximum volume of the system should be defined on the input side and should in fact be set using an attenuator between the audio board and the amplifier. It should not be physically possible to raise the system volume past the point where your own chosen speakers blow. Every system is different, every speaker choice is different, and so a software solution is not the correct choice for deciding a sane maximum output volume for your specific setup. That should be done in hardware instead before the pack is sealed up.

I myself have an attenuator between my audio board and amplifier to clamp the audio level. This is the correct way to handle amplifier input, and is how the DROK amplifier that we used before the GPStar version did it as well.

I am not going to close this issue immediately, but I will discuss it with the other devs to see what they think.

nomakewan, 2025-05-13:
Per discussion with other developers, indeed the proper solution to preventing damage to hardware is by using a hardware limiter (such as attenuating the input to the amplifier, just as the knob on the DROK amplifier did).

One of the most important considerations for this is that ostensibly such an option would be in the EEPROM, but the EEPROM is not initialized when the system is freshly-installed and powered up, and on top of that the EEPROM is wiped clean if a firmware update alters the structure of the objects saved in the EEPROM (it does this for safety so that corrupt values are not loaded after a firmware update).

Please do use a hardware solution to limit the audio level going into the amplifier.

closed 2025-05-13 by nomakewan