gpstar81/GPStar-proton-pack #403: Afterlife and frozen empire outer cyclotron lights inop.

seattlebuster, 2024-10-27:
### What actions were attempted?

Flashed to latest firmware on pack and wand. Updated sounds.   Now outer cyclotron lights only work on 1984 and 1989.  Afterlife and FE won’t work.  I checked my eprom that I have the correct stock led option for number of leds.  Super weird.

### What actually happened?

Cyclotron lights only work on 84 and 89 modes.

### What was expected?

Normal operation of lights from the movie .

### Firmware Version

Latest as of today.

### Mode in Use

Mode Original (Prop)

### Notable Hardware

Stock, plus inner cyclotron ring. 

### Homework Completed

- [X] I did my part!

seattlebuster, 2024-10-27:
Also updated esp32 firmware. All configs via WiFi show afterlife settings and proper led counts. Odd

seattlebuster, 2024-10-27:
I am also on the original setup using a mega board and wav trigger.  Wand is using gpstar board and wav trigger.  All cyclotron lights worked on previous firmware. Works in 84/89 mode, just not afterlife or FE.

nomakewan, 2024-10-27:
When you say "previous firmware" which specifically are you referring to?

seattlebuster, 2024-10-27:
I believe 5.3. I don't remember. But I think it was from August .

nomakewan, 2024-10-27:
Very strange; I am unable to replicate your report. Additionally without knowing what the previous firmware was, it's hard to say what actually changed. A firmware several months old was before a lot of changes were implemented, such as the change to default to the Frutto upgrades if you don't have a GPStar Lite kit and your EEPROM is on all defaults, and the change in how the ramp up/down outer cyclotron ring simulation works.

Since you say you updated the ESP32 and have it installed, I suggest you make sure that all the LED settings are correct for your setup, and that you then change another setting such as disabling the ring simulation and then save settings and commit to EEPROM. Once done, power cycle the pack. Does it work again?

seattlebuster, 2024-10-27:
I'll need to check when I get home. No idea on the pinout, I might have switched it over a long time ago. The lid lights worked on the last firmware.   So odd that it works in 84/89 mode. 

I'll keep checking,  I should switch my main board out to the gpstar. I was one of the early folks to do the hardwire method a couple of years ago.   Lots of soldering ahead to swap to the new board.  

I'll upload screenshots of my packs UI on the esp32 when I get home.

nomakewan, 2024-10-27:
The reason I ask about the pinout is that there _is_ a bug if I simulate removing the cyclotron lid. But in fact the bug is that the 84/89 outer cyclotron lights up at all. It shouldn't, but it does due to an oversight in how the 84/89 outer cyclotron code works (it doesn't check for the lid status when turning on/off LEDs, only for fading them). In fact, if I disable "Center LED Fade Effect" in the Proton Pack settings, I can replicate your bug report perfectly (84/89 works fine, AL/FE fails to light up).

This, coupled with the fact that there was indeed a change in how the pin for the cyclotron lid detection works, makes me think that perhaps that is the culprit. But we will find out shortly. :)

seattlebuster, 2024-10-27:
I'd rather switch the pinout so that it works with future firmwares.  I'll check for sure.   I don't necessarily want to replace the board since I cut all of the connectors to make it work on the mega shield. That would be a ton of soldering

seattlebuster, 2024-10-27:
I have a wire connected to d51.  So where should I place it now?

nomakewan, 2024-10-27:
> I have a wire connected to d51. So where should I place it now?

You should connect it to D43.

seattlebuster, 2024-10-27:
That was it! Forgive me lol.  I did not get the memo of the change.

I really need to move over to the gpstar board.

closed 2024-10-27 by seattlebuster

nomakewan, 2024-10-27:
Glad it was something simple! And it inspired me to look into the cyclotron code some more and patch up some stuff related to the lid detection, so all good!