gpstar81/GPStar-proton-pack #404: [Bug]: Can't interact with ESP32 interface

Toy203, 2024-10-27:
### What actions were attempted?

I can finally connect to the ESP wireless network with my Android phone! Awesome! Unfortunately, i'm unable to interact with the option. I can change pages but no button works. The homepage status is also showing wrong values i believe.

[image]

It is the same from phone or from PC access.

### What actually happened?

Same result on beta 6 and 5.4.2 RC2 at the moment. I've flashed every piece of the equipment twice (on each version) to be sure.

### What was expected?

I've flashed the ESP with the new beta files through USB initially as instructed, when 6.0 became a thing. Worked fine on 5.4 so far.

### Firmware Version

6.0.0 beta

### Mode in Use

Super Hero / Video Game

### Notable Hardware

ESP32, no attenuator

### Homework Completed

- [X] I did my part!

nomakewan, 2024-10-27:
To save Dustin some time, please connect on your desktop and then open your browser's web console and see if there are error messages displayed in the console. If so, please post them here.

Toy203, 2024-10-27:
Sure, will have a look as soon as possible.

DustinGrau, 2024-10-27:
Yes, open the developer console and then refresh the original page. You wanna look at the console to see if there's any JavaScript errors reported, and then look at the network page to see if any files did not download.

Toy203, 2024-10-27:
I don't see much errors aparto for some favicon message:
[image]

I can enter options menu but ESP reports it can't read pack status or wand.
[image]
[image]

Toy203, 2024-10-27:
Here's  a multiproces dump of me loggin in the home page, trying Music settings, Pack on button and getting into Options page.
[Uploading console-export-2024-10-27_17-22-55.txt…]()

DustinGrau, 2024-10-27:
OK everything here is pointing to the ESP is not communicating with the pack at all. Are you sure that you've powered everything down then back up at the battery after loading the latest firmware? Do you have the blue light on the device that indicates successful pairing? The light on top of the attenuator will now glow purple while it's still waiting for a connection, so if you have that, what color is it?

Toy203, 2024-10-27:
I did power cycle it a few times, it worked fine before the update and i didn't change anything hardware-wise. My pack is closed up so i have no clear vision of the leds on the ESP board unfortunately.

nomakewan, 2024-10-27:
> Here's a multiproces dump of me loggin in the home page, trying Music settings, Pack on button and getting into Options page. Uploading console-export-2024-10-27_17-22-55.txt…

Looks like this did not attach properly.

For the console, you'll see a "No" icon in the upper-left. Clicking it clears the console. While looking at the "Pack Status" page, clear the console so that nothing is in it, then refresh the page. That should give a clean list of all errors specific to loading the pack status.

Toy203, 2024-10-27:
> Looks like this did not attach properly.
>
> For the console, you'll see a "No" icon in the upper-left. Clicking it clears the console. While looking at the "Pack Status" page, clear the console so [...]

here it is.
[Uploading console-export-2024-10-27_17-30-50.txt…]()

DustinGrau, 2024-10-27:
On the network tab, you should see somewhere in there an initial status request for data. If that reports back that the wand is disengaged, or if looking at the main screen nothing changes when you turn the pack on or off, then that's a connectivity issue with the pack. 

Are you still running all v6 software at this point?

nomakewan, 2024-10-27:
That is patently bizarre. There is a bunch of stuff there that should not be there. I just tried in both Firefox and Chrome on 5.4.2 (pack and ESP32, no wand connected) and all I get in both browsers is:

```
Attempting to open a WebSocket connection... [index.js:46:11](http://protonpack_n1.local/index.js)
Connection opened [index.js:63:11](http://protonpack_n1.local/index.js)
```

Not that it should matter, but rather than accessing via IP address, have you tried accessing via the automatically-generated web address? It's in the form of `yourSSID.local`. So for example, my ESP32 wifi network is named `ProtonPack_N1` so my web address is `ProtonPack_N1.local`.

Toy203, 2024-10-27:
> On the network tab, you should see somewhere in there an initial status request for data. If that reports back that the wand is disengaged, or if looking at the main screen nothing changes when you [...]

It does change when i turn the pack on and reports different firing modes etc correctly. Just the voltage is at 0.0 from what i can see. Every other status (power cell on, wand safety...) is reported correctly. 
Yes i'm still on the latest 6.0.0 .
Tried to connect to .local. successfully but no changes. It also doesn't read music tracks from the SD cards, it fails to load the track list at all:
[image]
Dunno if that might help.

nomakewan, 2024-10-27:
To be clear, this means that when connecting to the `.local` version your console output is exactly the same? It shouldn't be. Could you check to be sure?

Toy203, 2024-10-27:
> To be clear, this means that when connecting to the `.local` version your console output is exactly the same? It shouldn't be. Could you check to be sure?

Sorry no, the functionality (or lack thereof) is the same. I'll upload the .local. Dump in a sec

Toy203, 2024-10-27:
Tried accessing .local with firefox (txt file ) and edge (log file), in case any of my privacy extensions on FF would interfere.

[console-export-2024-10-27_18-29-5.txt](https://github.com/user-attachments/files/17534828/console-export-2024-10-27_18-29-5.txt)

[protonpack_toy.local-1730050247111.log](https://github.com/user-attachments/files/17534830/protonpack_toy.local-1730050247111.log)

DustinGrau, 2024-10-27:
So just a nitpick on the terminology here, it looks like you can access the ESP32 and its web interface just fine. The problem appears to be some communications are not working 100%. You seem to get some information from the pack, but not everything works, is that correct? 

To be specific, are you getting information from the pack as you interact with it? But if you try to open one of the preference pages that reports that it can't get the information and says that the pack isn't attached, is that correct?

What I'm trying to identify is whether you are receiving from the pack, but not able to query it directly meaning the transmit is not working. And if that's the case, I would very much suggest opening things up and checking that your transmit and receive wires are all properly seated in their respective devices. Unlike ethernet or anything else that works on an acknowledgment basis, these devices truly can have unidirectional information sent. So the pack could be transmitting and the ESP 32 receiving, but the opposite may not be true due to physical fault most likely.

Toy203, 2024-10-27:
I can indeed access the ESP32 Web interface, but all the pack controls aren't doing anything. Can't turn pack on off, vent or play music, or change volume. I can't change any preferences as well. I might be able to open up the pack again after Halloween, I just opened it yo update ESP32 firmware with the new allocation table.

Again, everything was working fine on 5.4.1, I should try to downgrade and see if it changes something I guess.

nomakewan, 2024-10-27:
> Tried accessing .local with firefox (txt file ) and edge (log file), in case any of my privacy extensions on FF would interfere. [...]

Your Edge log looks exactly like my Firefox and Chrome logs--totally normal, nothing errant. The Firefox log does appear to be related to your privacy extensions, though there's an odd outlier error regarding the lack of favicon. That's something we can actually fix for 6.0.0 if we feel like it since we'll have enough space to store one.

However, I then went ahead and disconnected my serial data and found that no errors appeared in the log, so it looks like at some point the web diagnostics were either changed or removed for this particular feature. That being said, I agree with Dustin based on this information that this appears to be an issue with serial data misalignment between the Pack and ESP32.

Toy203, 2024-10-27:
Well it was PEBCAK all over again. I used the damn "standalone" firmware instead of the regular one. Works as expected with the CORRECT firmware lol.

closed 2024-10-27 by Toy203

DustinGrau, 2024-10-27:
Well, I'd call that a successful bench test

nomakewan, 2024-10-27:
Don't feel bad! As a result of this issue we now have an idea for hosting a favicon for the web UI, and it got me off my duff to update some favicon/manifest stuff with one of my own websites!

Glad it was something simple.