gpstar81/GPStar-proton-pack #406: [Bug]: System volume decreases during venting

DustinGrau, 2024-10-28:
### What actions were attempted?

Manually trigger a vent sequence or fire a stream long enough to go into overheat.

### What actually happened?

This one I noticed myself so logging a bug to make sure I remember to revisit it, but also inform others that it might be occurring. While either manually venting or entering an overheat sequence I noticed the system overall volume being reduced without any interactions on my part. I could see the % change on the web UI so it wasn't just me hallucinating or thinking it was something else. I manually had to turn up the volume after the venting to get it back to normal.

### What was expected?

Did not expect volume to randomly change without interaction.

### Firmware Version

5.4.2 (technically 6.0 Beta 2)

### Mode in Use

Super Hero / Video Game

### Notable Hardware

Running all the latest version boards, modified Wasteland smoke kit via relay.

### Homework Completed

- [X] I did my part!

nomakewan, 2024-10-29:
Looking into this, there are only two possibilities.

1. Somehow, the check [HERE](https://github.com/gpstar81/GPStar-proton-pack/blob/v6.0.Preview/source/NeutronaWand/src/main.cpp#L9844) is going wonky-doodle. This is impossible while firing, but since I do not know how you are initiating your overheat sequences out in the field I cannot discount it. The check will prevent this volume control from working if _either_ of the right-hand switches are in the 'on' position. However, I bring it up because it works by holding down Intensify, and holding down Intensify is what does the manual overheat sequence in Mode Super Hero as well (albeit with the bottom-right switch on and the top-right switch off).

2. Somehow, the rotary encoder on the Attenuator is going wonky-doodle. There are no guards on the rotary encoder's volume control other than if the wand is currently firing _and_ the current multiplier is greater than 2 (at which point it will function as the stream attenuator rather than the volume control). This can be seen [HERE](https://github.com/gpstar81/GPStar-proton-pack/blob/v6.0.Preview/source/AttenuatorESP32/include/System.h#L350). This means that when firing and the multiplier is 0~2, it will continue to act as a volume control. It will also act as volume control while the ribbon cable alarm is going off, and while overheating, and while venting, and while wand mashing, and..............you get the point.

Personally 2 seems more likely. Perhaps some more guards need to be put in place on the volume control, and the actual functionality of the rotary encoder itself should be inspected to make sure no errant signals are being passed along as turns.

nomakewan, 2025-04-21:
Just pushed a commit for 5.4.4 which resolves 2 in the post above. Honestly not sure why we didn't do it sooner, but it's done and I'm going to call that the solution for this bug. :)

commit 01e4962d (2025-04-21): Attenuator, Lite fixes

Adjusts GPStar Lite overheat lock to come on sooner to account for the wider firing check window, and makes it so that Attenuator volume controls cannot function if the user is firing their Neutrona Wand, which resolves #406
closed 2025-04-21 by nomakewan