gpstar81/GPStar-proton-pack #408: [Bug]: Inner Cyclotron way too fast

nomakewan, 2024-10-31:
### What actions were attempted?

Turn on the pack with the lid off.

### What actually happened?

Inner cyclotron rotation speed is insanely fast.

### What was expected?

Inner cyclotron rotation speed should be slow enough at idle to perceive a single dot moving in a circle.

### Firmware Version

5.4.2/6.0.Preview

### Mode in Use

Super Hero / Video Game

### Notable Hardware

35 LED Inner Cyclotron Ring.

### Homework Completed

- [X] I did my part!

Toy203, 2024-10-31:
Honestly thought that was the wanted behavior on this one. It looks cool on mine lol.

nomakewan, 2024-10-31:
It wasn't intended as far as I'm aware. If no one reported the change I wonder when it actual happened...

Toy203, 2024-10-31:
Here's mine on 6.0
https://youtu.be/7b0hGVFbi6I

I guess the plexiglass lid could help a bit with the effect, but honestly it looks fine to me. Ramp up and down works well and I like the overall effect.

nomakewan, 2024-10-31:
> Here's mine on 6.0 https://youtu.be/7b0hGVFbi6I

Which LED ring is this?

Toy203, 2024-10-31:
The OG 35 ring, afterlife settings.

nomakewan, 2024-10-31:
Interesting. I too have the OG 35 ring, and while it works fine in 84/89, AL/FE are insanely fast to the point where the LED pattern is broken up pretty badly. I'll have to do some experiments, like introducing the Frutto inner panel into the circuit to see if that has any effect. I've already confirmed that changing the _outer_ cyclotron configuration has zero effect on the behavior of the inner cyclotron, so there's that at least.

Toy203, 2024-10-31:
Don't get me wrong, I can see the pattern is broken on full rotation speed, leds don't sequentially turn on from what I can see. I just thought it was some kind of trail effect in place or something.

nomakewan, 2024-10-31:
Nah, it's the actual rotation effect outpacing the FastLED refresh rate. This allows the LEDs to go "faster" but since they're not being refreshed that quickly, LEDs get skipped over.

closed 2024-10-31 by nomakewan