gpstar81/GPStar-proton-pack #409: [Feature]: Quick Theme Change for Web UI

DustinGrau, 2024-11-04:
### What would you like to add in terms of software changes?

Currently you must select the 4th tab on the UI, then Proton Pack settings, change the theme via dropdown, then save and optionally commit preferences. Having used this ability quite often, and for those who prefer this over the menu system via the wand, I realize this would be nice to have a more accessible level. At least in the case of the toggles which are directly exposed via the pack controller (vibration, smoke, theme, cyclotron direction) it's worth considering making these more available.

### Would this request involve any specific hardware?

Nope

### Homework Completed

- [X] I did my part!

nomakewan, 2024-11-04:
Possible integration with #377 as well?

nomakewan, 2025-09-25:
This is now implemented in the upcoming firmware that includes GPStar II support.

closed 2025-09-25 by nomakewan