gpstar81/GPStar-proton-pack #412: [Feature]: Inverted Cyclotron Pan lights? Possible or not?

The-Y33TER, 2024-12-16:
### What would you like to add in terms of software changes?

I'm working on a custom build and I'm going for a very different look than anything I've seen before, which is the cyclotron Pan lights being inverted, with 3 lenses being on and one being off, in the AL/FE modes. It seems like a simple addition, but I am a little bit worried about amperage draw (Saw in a different post that the fully lit cake lighting draws a whopping 700ma) I'm aware it's a tad unorthodox, but this would absolutely make my pack. Thank you!!

(Also very happy there's now a RGB vent/gun top arcolectric indicator option in the store, very cool!)

### Would this request involve any specific hardware?

Cyclotron Pan LED boards, no new hardware ideally :>

### Homework Completed

- [X] I did my part!

nomakewan, 2024-12-17:
It is possible but I'm not sure yet how we would get that option into the menu. 6.0 will be a pretty major overhaul including a huge change to all the menu systems, so I'll keep this in mind for that.

I'm happy about the RGB vent/top light too--it's always bothered me that you couldn't get accurate colors for an 84 wand build. Now we can!

The-Y33TER, 2025-01-14:
Another question for once 6.0.0 is released, would it be possible to have an option to invert the level 1-5 red to orange shift? I feel for my gear it makes more sense to turn more red at higher levels than orange (implying the movie gear is standard, while my pack has an overclocked cyclotron) 

I don't wanna be stuck at the lowest power level 100% of the time, to maintain a red stream, since I'm gonna have level 4 as my regular firing (smoke off) with level 5 as the mode with overheat. 

Yes, I know the menu is super crowded, and I'm sorry. This does seem like something that should be an option though.

nomakewan, 2025-01-14:
It's already inverted for GB2 mode, and we're already supporting templates in 6.0.0 (that is, you can create a custom template that mixes-and-matches features from all year modes).

Let's keep an issue page to a single topic if we can, it keeps things more focused and easier to manage.

The-Y33TER, 2025-01-15:
Sorry, didn't wanna clutter with making another issue, since I've created a few already. Super excited for my kit and 6.0.0!!
Is there a rough release date for 6.0.0 or just "when it's done it's done"

nomakewan, 2025-01-16:
Done when it's done. 6.0.0 isn't just moving menus around and adding a template feature, it's also redoing a lot of LED-related code to make it more efficient. That process hasn't even started yet in earnest, so it'll take a while.

Right now focus is instead on 5.4.3 to support a few new LED parts that were sold, fix a few minor bugs, and hopefully fix some issues with the Lite kit that users have reported.